ARM-software/CMSIS_5 #1047: NN_ROUND() issues in 16bit ops

majianjia, 2020-11-20:
I met an issue of 16 bit NN producing bad results when I tried to use `round()` in NN calculation using `ARM_NN_TRUNCATE` (undefined ARM_NN_TRUNCATE)

https://github.com/ARM-software/CMSIS_5/blob/e42a76a56841fed4dbd182221cc7f3ade753acc9/CMSIS/NN/Include/arm_nnsupportfunctions.h#L652

change `#define NN_ROUND(out_shift) ( (0x1u << out_shift) >> 1 )` to 
`#define NN_ROUND(out_shift) ((q31_t)( (0x1u << out_shift) >> 1 ))` fixed my issue. 

Also, it doesn't seem right when it adds to a negative number. Should the round be round down in negative number? In most ops, only add this to the bias which means it is always positive. 

More information will be provided when I figure it out. Thanks.

felix-johnny, 2021-08-13:
@majianjia Apologies for the delay. Would you mind making a PR for the fix?

majianjia, 2021-08-13:
@felix-johnny let me reminding myself what is this about.
Happy to, thanks :p

felix-johnny, 2021-08-13:
Great , thanks!

majianjia, 2021-11-02:
close issue as the PR is merged. 
Thanks

closed 2021-11-02 by majianjia