ARM-software/CMSIS_5 #1049: SVDconv and documentation of __IM/__OM/__IOM

sebastienmasson, 2020-11-25:
SVDconv generate a doxygen entry from those three macros as follows (just after end of group "Configuration_of_CMSIS")

```
#ifndef __IM                                    /*!< Fallback for older CMSIS versions                                         */
  #define __IM   __I
#endif
#ifndef __OM                                    /*!< Fallback for older CMSIS versions                                         */
  #define __OM   __O
#endif
#ifndef __IOM                                   /*!< Fallback for older CMSIS versions                                         */
  #define __IOM  __IO
#endif
```

It seems those documentation entries are not properly supported by doxygen (1.8.17) and results in the following output.

[image]

[image]

JonatanAntoni, 2020-11-25:
Hi @sebastienmasson,

thanks for pointing this out.
The pasted generated code looks like having the Doxygen comments in the wrong line. 

Can you please try to fix the code to look like this, manually:
```
#ifndef __IM
  #define __IM   __I                                    /*!< Fallback for older CMSIS versions                                         */
#endif
``` 

The Doxygen comment marker `/*!<` denotes a comment applied to the preceeding definition.
Actually, I am not sure if this comment should be a Doxygen one at all.

Cheers,
Jonatan

sebastienmasson, 2020-11-25:
```
#include "core_cm0.h"                           /*!< ARM Cortex-M0 processor and core peripherals                              */
#include "system_MCUxxx.h"                      /*!< MCUxxx System                                                             */

#ifndef __IM
  #define __IM   __I	/*!< Fallback for older CMSIS versions                                         */
#endif
#ifndef __OM
  #define __OM   __O	/*!< Fallback for older CMSIS versions                                         */
#endif
#ifndef __IOM
  #define __IOM  __IO	/*!< Fallback for older CMSIS versions                                         */
#endif
```
[image]

[image]

JonatanAntoni, 2020-11-25:
Okay, another issue seems to be the Doxygen comments on the `#include` lines. I guess these should be normal C comments instead.

flit, 2021-01-05:
I would completely exclude the `__IOM` and related definitions from doxygen. And definitely the include directive comments should not be doxygen-style.

sebastienmasson, 2021-02-10:
Do you still need a feedback on this?