ARM-software/CMSIS_5 #1050: SVDconv - Customizing the output

sebastienmasson, 2020-11-25:
Is there a mean to customize the output doxygen tags?

In particular "@file" : I would like it to be a single file name with no path.

However, it would be convenient to let the user the customize them all.  It could be done through a xml file for instance.

Thank you :-)

JonatanAntoni, 2020-11-25:
Hi @sebastienmasson,

no, currently the template for generated header files is hard coded in SVDConv.

Cheers,
Jonatan

sebastienmasson, 2020-11-25:
Ok

I would also like to emphasize it would be very useful to be able to  customize the ```\addtogroup xxx``` statements.
Doing so, we could easily include documentation for several <device>.h within the same doxygen output.

Doxygen is already difficult enough ;-)