ARM-software/CMSIS_5 #1051: Can spi multi slave driver use on diffefrent spi ports?

Pupil2013, 2020-11-25:
Some chips have more than one spi port,how can the driver used on two or more spi ports?
I have two spi ports ,they all have multi slaves,so how can the multi slave drive used on two ports?

Another Question
I'm using the spi driver on multi slaves,but the parameters can't be changed on different slave devices,like the data bits,
I use Control function to change the parameter on different slave,but it seems not changed.Am I using it right?

Pupil2013, 2020-11-25:
Init the driver first:
`Driver_SPI2.Initialize(NULL);
Driver_SPI2.PowerControl(ARM_POWER_FULL);`
then first slave to config the spi para:
`ptr_Sx1278_Spi->Control(ARM_SPI_MODE_MASTER | ARM_SPI_CPOL0_CPHA0  |
                                                  ARM_SPI_DATA_BITS(8) |
                                                  ARM_SPI_MSB_LSB      |
                                                  ARM_SPI_SS_MASTER_UNUSED,
                                                  SX1278_SPI_BUS_SPEED);`
then when anogher:
`ptrLCDSPI->Control(ARM_SPI_MODE_MASTER | ARM_SPI_CPOL0_CPHA0  |
                                             ARM_SPI_DATA_BITS(9) |
                                             ARM_SPI_MSB_LSB      |
                                             ARM_SPI_SS_MASTER_UNUSED,
                                             LCD_SPI_SPEED);`

the data bits on the last one is still using 8 bits

JonatanAntoni, 2020-11-25:
Hi @Pupil2013,

It might be easier to answer when raising different questions with separate issues.

Let me quickly address your first question: Basically, each port is handled with a dedicated driver instance. If you have two SPI ports, you should have two SPI-Driver instances in your system.

The second question needs to be inverstiaged in detail. Can you give us some more details about the devices and drivers you are using? I.e. which vendor? Which device familiy pack and version?

Cheers,
Jonatan

Pupil2013, 2020-11-25:
Thanks @JonatanAntoni 
Can I use multi slave wrappers on different spi ports,I can't find where to config that on different ports

Pupil2013, 2020-11-25:
Some details about the second question:
I'm using STM32L476RCT6 ,the cmsis spi driver verison is 2.3.0

Pupil2013, 2020-11-25:
Sorry @JonatanAntoni 
About the second question ,I checked my code again ,I think it's my fault,I'm using the spi port on the wrong way

RobertRostohar, 2020-11-25:
As mentioned before, if you have multiple SPI ports then you need an SPI driver for each port.

If you are looking to support multiple slaves on a single SPI port then you need a multi-slave wrapper on top. Such a wrapper is available as part of the CMSIS-Driver pack.

Take a look here: https://arm-software.github.io/CMSIS-Driver/General/html/driver_SPI.html

Pupil2013, 2020-11-26:
Thanks @RobertRostohar
So the multi-slave wrapper can't be used on two spi ports simultaneously。

RobertRostohar, 2020-11-26:
The wrapper sits on top of a **single** physical SPI driver and exposes **multiple virtual** SPI drivers (one for each peripheral connected to the same physical port). Please take a look at the documentation link provided earlier for more details.

Each physical SPI port however requires its own SPI driver (regular driver and not the virtual driver mentioned above).

So a single wrapper can't be used on two SPI ports simultaneously. A wrapper per SPI port would be needed.

Pupil2013, 2020-11-26:
Thank you ,I got it now

closed 2020-11-26 by Pupil2013