ARM-software/CMSIS_5 #1053: "Convert n-bit Integer value" functions may be better labelled "Fixed point"

hachi, 2020-11-26:
I am not well versed in the topic of number formats, but I believe the section DSP/Reference/Support Functions is misleading in the label for the 32/16/8 bit "convert value" function.

I believe they would be better labelled as "fixed point" instead of "integer". I do wonder why there are no functions (that I can see) for converting integers to Q type numbers.

If these functions can actually be used for the conversion of n-bit integer values I would really suggest a small example of some kind.

Thank you for this library and the maintainership of it.

christophe0606, 2020-11-26:
@hachi You are right. The documentation is not correct and should not use "integer". Thanks for reporting it.

Q31/Q15/Q7 type numbers are always < 1 in absolute value. So, we can't convert an integer to those formats.

But we can typecast an integer to those values if the integer is already representing a fixed point so there is no need to have a specific conversion function.

christophe0606, 2021-09-30:
Corrected in commit 18bfe3338a722117ad079ea47d7e6224fb6b2b3e

closed 2021-09-30 by christophe0606