ARM-software/CMSIS_5 #1059: Update documentation for Packs 1.7.0

ilg-ul, 2020-12-01:
While processing the CMSIS Packs in [Eclipse Embedded CDT](https://projects.eclipse.org/projects/iot.embed-cdt) (formerly GNU MCU Eclipse), I get 4 messages:

```
Schema version "1.7.0" not recognised, package Keil.NUCLEO-L552ZE-Q_BSP.pdsc ignored.			-	CMSIS Packs
Schema version "1.7.0" not recognised, package Keil.STM32L552E-EV_BSP.pdsc ignored.			-	CMSIS Packs
Schema version "1.7.0" not recognised, package Keil.STM32L562E-DK_BSP.pdsc ignored.			-	CMSIS Packs
Schema version "1.7.0" not recognised, package Keil.STM32L5xx_DFP.pdsc ignored.			-	CMSIS Packs
```

According to the current [documentation](https://www.keil.com/pack/doc/CMSIS/Pack/html/pack_revisionHistory.html), the latest version is 1.6.3.

Please update the documentation to reflect the new 1.7.0 version, so I can further update the parser in Eclipse.

JonatanAntoni, 2020-12-01:
Hi @ilg-ul,

in fact the schema version has already enhanced to 1.7.1, see https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Utilities/PACK.xsd. This is not properly reflected in the version history included with the documentation.

Thanks,
Jonatan

ilg-ul, 2020-12-01:
> not properly reflected in the version history

That's the purpose of this ticket, to update the documentation.

Me, as a consumer of CMSIS, I don't check internal files, I check the top pages, and in this case missed your updates.