ARM-software/CMSIS_5 #1061: STM32H7 CMSIS drivers

shirocro, 2020-12-02:
I am not sure is it right place to ask this so my apologies at the beginning. I want know who is working on CMSIS.packs which could be downloaded in Keil?
I have a question about CMSIS emac driver for STMH7 series of mcu-s. This ethernet driver is barely implemented with basic functionality, there is no timecode implementation etc.. I am just curious when should I expect those implementations or maybe I can contribute for it.

RobertRostohar, 2020-12-02:
I'm not sure what you mean by 
>This ethernet driver is barely implemented with basic functionality

The driver is fully functional (including checksum offload by hardware). It does lack only support for Precision Timer which is however a feature not often used by network stacks.

I suggest you contact support at developer.arm.com/support.

shirocro, 2020-12-02:
Yeah, you are correct, but I need support for Precision Timer, and for any other series of STM32F7, F4... is implemented.

shirocro, 2020-12-02:
Is somewhere github repository of specific device CMSIS pack, for example STM32H7xx is what I am interested currently?

RobertRostohar, 2020-12-02:
The Device Family Pack for STM32H7xx (Keil:STM32H7xx_DFP) is not on github at this point.