ARM-software/CMSIS_5 #1068: incompatible type error when building cmsis-5.7.0 using gcc

shiyuanren, 2020-12-11:
arm_biquad_cascade_df2T_f32.c:92:19: error: incompatible type for argument 1 of '__arm_vshlcq_s32' 
include\arm_mve.h:6925:29: note: **expected 'int32x4_t' but argument is of type 'f32x4_t' {aka 'float32x4_t'}**

basicmathfunctions\arm_abs_f32.c
arm_helium_utils.h:233:14: error: incompatible type for argument 2 of '__arm_vldrwq_gather_shifted_offset_s32'
  233 |     vecTmp0 = vldrwq_gather_shifted_offset_s32(sqrtTable_Q31, vecIdx);
      |              ^~~~~~
      |              |
      |              q31x4_t {aka int32x4_t}
note: **expected 'uint32x4_t' but argument is of type 'q31x4_t' {aka 'int32x4_t'}**

How to solve this problem？ 
Thanks

christophe0606, 2020-12-14:
@shiyuanren Unfortunately, the library has not yet been tested with the gcc including Helium support. It was not available when 5.7.0 was released.

We are working on it but I can't communicate any schedule.

In the meantime you can add typecasts where needed and in some case replacing vld1q with vld1q_f32 to make the type explicit if the compiler can't infer it.

christophe0606, 2020-12-14:
@shiyuanren  You cna also try to add the option  -flax-vector-conversions when compiling

closed 2021-02-01 by shiyuanren