ARM-software/CMSIS_5 #1079: Unify the naming style of CMSIS

dsyx, 2020-12-22:
Currently, each sub-module of CMSIS uses a different naming style (for example, in DSP, function names use lowercase and underscores, but in RTOS, function names use camel case.). A unified naming style can provide better readability and ease of use.

ilg-ul, 2020-12-22:
+1 for lowercase and underscores.

This is the style used by the C/C++ standard libraries.

JonatanAntoni, 2020-12-22:
Hi @dsyx,

basically you're right that having a common coding style is nice to have for a project.

Please be aware that CMSIS is *not* a single project, even if maintained in a single repository. Each of the "sub-modules" are driven by separate teams and have their individual, long lasting histories (many years before the integration into CMSIS 5 repostitory). 

Massively renaming all this is not an option from my point of view. The effort needs serious investment. And renaming a lot of stuff is a breaking (hardly backward compatible) change which will affect lots of user projects. For this reasons I don't expect this will be addressed.

Cheers,
Jonatan

dsyx, 2020-12-22:
Hi @JonatanAntoni ,

Yes, renaming on a large scale is indeed a time-consuming and laborious act.

But CMSIS is a standard specification, and it will be used by a large number of vendors and developers. If CMSIS can't unify the coding style well, various styles of code will appear in the vendor's SDK or developer's project code, which will make the SDK/project difficult to read and maintain.

So, it might be worth it.

Cheers,
Yaoxing

JonatanAntoni, 2020-12-22:
Hi @dsyx,

I admit this impact. We already have such a very diverse situation as most SDK vendors do have their own coding styles. I consider the impact CMSIS can have as less important. 

Breaking existing applications is much more worrying to me. And convincing all the different teams working around CMSIS to agree on a common coding style might be far more challanging. Both could be addressed in a next major version but not in a long lasting maintenance phase.

Cheers,
Jonatan

dsyx, 2020-12-22:
Hi @JonatanAntoni ,

Yes, I think this problem can be handled in CMSIS V6.

Cheers,
Yaoxing

GorgonMeducer, 2021-04-27:
I possible solution is :

1. create a version with a unified name style
2. create a wrapper layer for backward compatibility
3. declare deprecate for those old symbol names and set a time table for deprecation. 

What do you think?

Cheers,
Gabriel

JonatanAntoni, 2021-04-27:
@GorgonMeducer,

Sure, if we could come up with a general conclusion about coding style this might be an option. But I doubt CMSIS can have this impact at all. And if it could, we'd need to have the style guide first so that others could at least comment.

What we'd need to do:
1. Come up with a write-up about the recommended coding style.
2. Have an intensive review by all affected teams.
3. Create an detailed analysis what is subject to be renamed.
4. Conclude if that effort is ultimately worth it.

I am happy if someone wants to start with the first item.

Cheers,
Jonatan

dsyx, 2021-04-28:
Regarding the coding style, I recommend referring to the style of the Linux kernel in general and making slight modifications in details. The advantages of this are:

1. It can save a lot of time for writing style guides.
2. LInux kernel style is similar to standard C style.
3. Most ARM developers are also Linux developers.

What do you think?

Cheers,
Yaoxing

JonatanAntoni, 2021-04-28:
Thanks @dsyx,

> 1. It can save a lot of time for writing style guides.

+1

> 2. Linux kernel style is similar to standard C style.

I am not familiar with this. Can you provide a link, please? Would you simply take it as-is? Or shall we take it as a baseline but adopt?

> 3. Most ARM developers are also Linux developers.

Personally, I am not (at least not more than on a basic level). CMSIS is used for bare metal, deeply embedded applications which are totally different to applications on rich OSes like Linux. Windows based computers are commonaly used in industrial grade embedded development. 

> What do you think?

I do absolutely prefer using an established style instead of creating yet another custom one.
Can we have a write-up of the proposed style, ideally with certain examples?

Cheers,
Jonatan

dsyx, 2021-04-28:
Hi @JonatanAntoni,

I think the Linux kernel style is very similar to the standard C style. Because I did not find a standard C coding style specification, so I proposed to use the Linux kernel style. If you can get the standard C coding style, it is of course a better choice.

> I am not familiar with this. Can you provide a link, please?

Please check: [Linux kernel coding style](https://www.kernel.org/doc/html/v4.10/process/coding-style.html)

Cheers,
Yaoxing

ilg-ul, 2021-04-28:
For inspiration you can also take a look at http://micro-os-plus.github.io/develop/naming-conventions/.

JonatanAntoni, 2021-04-28:
Lets consider the mbedOS guidelines as well:
https://os.mbed.com/docs/mbed-os/v6.9/contributing/style.html

It might make sense to have this closely related projects using the same style.

dsyx, 2021-04-28:
I think the mbedOS guidelines is good.

But I think two details in the mbedOS guidelines can be modified:

> Cases within switch are indented (exception from K&R).

1.  Do not indent the case, this is unnecessary and wastes 4 character space.

> Enumeration - the type name and values name - same naming convention as classes (for example MyNewEnum).

2. Enumeration classes and their values can use the class naming style. But the enumeration type should use the naming style of the structure, and the value of the enumeration type (the enumeration type does not limit the visible range of its value) should use the naming style of the macro.

What do you think?

JonatanAntoni, 2021-04-28:
@dsyx, for CMSIS we can have this decision. I doubt mbedOS team would aggree. I'll check the coding style preferences with other parties and report back.

GorgonMeducer, 2021-04-29:
@JonatanAntoni 

I have different thought about coding style. The coding style is different from the coding principle, it is just a set of habits someone has and enforced in the range the guy can control. It's nothing more than a style of hand-writing.

You can follow a perfect nice coding style and writing a totally bullshit code.
You can also find reliable and high-quality code written in mixed coding styles (just like CMSIS and other open-source projects you could find in Github).

So for what you said:

1. Come up with a write-up about the recommended coding style.
    yes, find someone you trust (or just yourself) propose a solution.

2. Have an intensive review by all affected teams.
    No need for an intensive review. Just enforce it, and it saves everyone's time. 

Cheers,
Gabriel