ARM-software/CMSIS_5 #1080: FFT Function Bit Reversal Parameter

DanSzaboStoneAge, 2020-12-30:
After discussion regarding optional bit reversal on a couple DSP forums, it would seem the only sensible reason to include a parameter for en/disabling bit reversal is because it isn't required for convolution operations, as long as the FFT of the impulse response is similarly bit reversed.  However, this implementation uses natural input/reversed output for both forward and inverse transforms.  This means that if bit reversal is disabled, the forward transform can't be reversed.

Was this intentional for some reason, or was this an oversight in the implementation?  Why even offer it as a parameter?  I am unable to think of an example of an application for which not using bit reversal would be advantageous with this implementation.  

Thanks much and happy new year!

christophe0606, 2021-01-04:
@DanSzaboStoneAge You are totally right.

The possibility of disabling the bit reversal is useful when:

- A linear processing is applied between the FFT and IFFT  (like convolution)
- The FFT and IFFT are in some way dual : bit reverse is applied to the output of the FFT and input of the IFFT.

In CMSIS-DSP, the bit reverse is applied at the output of both the FFT and IFFT.

So the use cases where it may be useful are rather limited : filtering of the spectrum and computing some energy after it the only use case I can imagine now.

CMSIS-DSP is an old library and unfortunately I don't know why some design choices were made .

DanSzaboStoneAge, 2021-01-04:
@christophe0606 Thank you for the response, makes good sense to me.  It may be worth updating the docs to reflect that, as I spent a bit of time chasing my tail on that one.  Something like "Always set the bit reversal flag to 1 unless you have a good reason, thar be dragons" but in more ARM-y type language.  I suppose you could create new IFFT implementations to reverse on the input side, but that might constitute a breaking change so it might not be well advised.  I personally think the benefits for linear operations would be worth it.  You'd probably get a slight performance bump on the IFFT as well, as you can put the conjugation of the twiddle factors into the source, instead of conjugating the whole of the input data prior to the radix call.  Thanks though, I am all straightened out.

closed 2021-01-05 by christophe0606