ARM-software/CMSIS_5 #1086: suppress pc-lint error 27: Illegal character (0x24)

connikullmann, 2021-01-05:
using pc-lint with files including cmsis_armcc.h or cmsis_armclang.h I get the following pc-lint errors:

#define __INITIAL_SP              Image$$ARM_LIB_STACK$$ZI$$Limit
#define __STACK_LIMIT             Image$$ARM_LIB_STACK$$ZI$$Base

C:\Keil\ARM\CMSIS\Include\cmsis_armcc.h/cmsis_armclang.h(119,0): Error 27: Illegal character (0x24)

Is it possible to suppress this message by adding /*lint -ident($) */ in future cmsis releases ?

GuentherMartin, 2021-01-11:
Hello @connikullmann,
is it possible to solve this problem by changing the used lint file?

We want to avoid tool-specific changes, in this case PC-Lint, in the CMSIS source/header files.
Some people use PC-Lint, others use Coverity or other static analysis tools.
In my opinion it is not a good solution to to add tool-specific filter to the code to prevent an analysis tool to generate a warning or an error.
Better would be to change the tool's configuration file.

GuentherMartin, 2021-02-04:
Hello @connikullmann,
Please add the Lint option `-ident($)` to your used Lint file.

If this answer is satisfying then please close this issue.

connikullmann, 2021-02-07:
solved

closed 2021-02-07 by connikullmann