ARM-software/CMSIS_5 #1087: No trailing whitespace requirement

flit, 2021-01-05:
The CMSIS source files, including Doxygen sources, need a standard formatting requirement regarding no trailing whitespace. I don't believe any CMSIS sources are missing the final newline, but that should be included as a formatting requirement.

I'll be happy to add an [`.editorconfig`](https://editorconfig.org) file to define the requirements, as well as clean up the existing sources.

We can include the existing indentation rules in the config. For most CMSIS files, it's 2 spaces. For NN it looks like 4 spaces.

GuentherMartin, 2021-01-14:
A practical proposal is to use .editorconfig files local to each component and have one common .editorconfig on top level.
In this way we have a hierarchy of .editorconfig, so there is a top-level config and then any components that have a differing style have their own. (btw, different format rules can be applied to filename patterns).
We just have to agree on the content of the top level .editorcomfig.
The simplest solution would be to keep it empty.

flit, 2021-01-16:
Having an empty top level is not a bad idea. That would avoid accidentally applying common rules to components that have a different formatting requirement. The main downside is duplication, but, for the most part, the config files should be set up once and remain unchanged.

I guess the next step is to understand what the formatting requirements are for each component. Is that already documented?