ARM-software/CMSIS_5 #1088: doc file excluded from Keil uVision workspace when part of a component

fred-r, 2021-01-07:
Hi,

I am currently facing an unexpected Keil uVision behaviour when using CMSIS pack components and clayers with CMSIS-Build (0.10).
I am working with a given application having:
- 1 application level clayer
- 1 target level clayer
This generates a .cprj file that can be opened with Keil uVision without any issue.

Nevertheless, if I write my application level clayer like this:
  <components>
    <component Cclass="Device" Cgroup="Application" Csub="Samples Common files" Cvendor="STMicroelectronics" layer="App"/>
  </components>
  <files>
    <group name="Doc">
      <file category="doc" layer="App" name="./readme.md"/>
    </group>
    <group name="Sample">
      <file category="sourceC" layer="Sample" name="./sample.c"/>
    </group>
  </files>

Then I get 2 groups in the uVision workspace (Doc and Sample) with the 2 files (readme.md and sample.c) being present.
This is fine.

But, if I write my application level clayer like this:
  <components>
    <component Cvendor="STMicroelectronics" Cclass="Device" Cgroup="Application" Csub="Samples Common files" layer="App"/>
    <component Cvendor="STMicroelectronics" Cclass="Device" Cgroup="Application" Csub="HAL Samples" Cvariant="CRC_UserDefinedPolynomial" layer="App"/>
  </components>
  <files>
  </files>

Relying on a component "CRC_UserDefinedPolynomial" described like this:
    <component Cclass="Device" Cgroup="Application" Csub="HAL Samples" Cvariant="CRC_UserDefinedPolynomial" Cversion="0.1.1" condition="STM32 HAL CRC">
      <description>CRC_UserDefinedPolynomial</description>
      <files>
        <file category="doc"    name="CRC/CRC_UserDefinedPolynomial/readme.md"/>
        <file category="header" name="CRC/CRC_UserDefinedPolynomial/sample.h"/>
        <file category="source" name="CRC/CRC_UserDefinedPolynomial/sample.c"/>
      </files>
    </component>

Then:
1. I do not have groups any more => OK, this I can understand
2. The readme.md file is not present in the workspace

In both cases the readme.md file is tagged as "doc".
So why is it removed from the workspace in the second case ?

Thanks & Regards,
Fred

fred-r, 2021-01-07:
1st code snippet :
"
<components>
    <component Cclass="Device" Cgroup="Application" Csub="Samples Common files" Cvendor="STMicroelectronics" layer="App"/>
  </components>
  <files>
    <group name="Doc">
      <file category="doc" layer="App" name="./readme.md"/>
    </group>
    <group name="Sample">
      <file category="sourceC" layer="Sample" name="./sample.c"/>
    </group>
  </files>
"

fred-r, 2021-01-07:
`  <components>
    <component Cclass="Device" Cgroup="Application" Csub="Samples Common files" Cvendor="STMicroelectronics" layer="App"/>
  </components>
  <files>
    <group name="Doc">
      <file category="doc" layer="App" name="./readme.md"/>
    </group>
    <group name="Sample">
      <file category="sourceC" layer="Sample" name="./sample.c"/>
    </group>
  </files>``

jkrech, 2021-01-08:
Hi Fred, 
if you want a copy of a component file being copied into your workspace you have to set the file tag attribute 
`attr="config"`
By default all specified files stay within the pack and are only referenced. Only configuration files are considered being "editable" and therefore they should be copied into the workspace and are treated as "project specific". Usually documentation files are considered read-only.
A CMSIS-Pack component can only have a single file of category="doc". If you take a look at the Manage Run-Time Environment Dialog in uVision, you will see that in the description column many descriptions are displayed as links. When clicking these links the doc file will be opened.
Does this answer your question?

Regards,
Joachim

fred-r, 2021-01-08:
Hi Joachim,

I think that the attr="config" is not what I need.
It is fine for me to have the doc file in readonly.

The point I do not get is:
- if I indicate my file explicitely in the clayer, it is part of the workspace
- if I indicate only a component listing this file, it is not part of the workspace

[doc_file_excluded_from_uVision.docx](https://github.com/ARM-software/CMSIS_5/files/5787193/doc_file_excluded_from_uVision.docx)

Thanks & regards,
Fred

jkrech, 2021-01-08:
Hi Fred,
sorry, "too close to the product" syndrom.
The simple answer is that uVision makes a difference between files being part of a 'Group' and a 'Component'. 
For components the project view only displays source files (C/C++/asm). You have probably already noticed that header files are not displayed either. The same is true for documentation. Component documentation is only accessible through the RunTime environment dialog.
It is worth noting that despite the fact that a file is listed in the project window, does not mean it is physically located somewhere in the project directory. If you hover your mouse over a file name in the project view it will display a tool-tip with the physical location. Similarly the project view does not list all files that are in the physical project directory.
I hope I got it right this time.

Regards,
Joachim

fred-r, 2021-01-08:
Hi Joachim,

you got it right and I agree with the difference between workspace and physical location of the files.
My point is : why a component should behave in a different manner ?
If a "doc" file is seen when it is in a group, why is it not seen when it is in a component ?
It is still a doc file in both cases (same taxonomy).

Thanks & Regards,
Fred

jkrech, 2021-01-08:
If I had the choice, I would not display documentation in the project view. Ultimately a design choice for the tool we should not get hung up about.

closed 2021-01-08 by jkrech