ARM-software/CMSIS_5 #1089: Is there a command line tool for managing the CMSIS packs ?

fred-r, 2021-01-07:
Hi,

I can easily trigger the pack manager tool by doing something like:
`start ${PackFile}`

What I would like to do is 2 things:

1. Install a CMSIS pack in my CMSIS_PACK_ROOT without opening the pack manager UI : I am looking for a cli.

2. Manage my local packs with a cli.
Today, in the pack manager UI, I can click on:
   'File->Manage Local Repositories...'
Then I canindicate a "local pack" so basically a folder with a pdsc file and the appropriate files.

What i'd like to do is to add a local repository thanks to a cli without using the UI.

Thanks & Regards,
Fred

flit, 2021-01-07:
See [pyocd](/pyocd/pyocd) and [cmsis-pack-manager](/pyocd/cmsis-pack-manager).

fred-r, 2021-01-08:
Thansk for the hint, I will give a try with cmsis-pack-manager.

Thanks & regards,
Fred

fred-r, 2021-01-13:
Hi @flit,

I cloned the cmsis-pack-manager github and did:
   pip install cmsis-pack-manager

Then I tried:
   python pack_manager.py

But this does not display any help and the documentation link seems to be broken.

Is it the proper way of using the pack manager ? (I see there are commands and subcommands but at the moement I do not know how to launch it, even the help text is not displayed).

Probably my way of calling the tool is very wrong ?

Thanks & Regards,
Fred

JonatanAntoni, 2021-01-13:
Hi @fred-r,

You might want to consider posting questions for cmsis-pack-manager to the according GitHub repo, i.e. https://github.com/pyocd/cmsis-pack-manager/issues.

Cheers,
Jonatan

fred-r, 2021-01-15:
Moving to cmsis-pack-manager github as suggested

closed 2021-01-15 by fred-r