ARM-software/CMSIS_5 #1094: Question about cbuildgen compose

fred-r, 2021-01-13:
Hi,

the documentation about cbuildgen compose indicates:

compose | Create new project | Create new   <ProjectFile>.cprj from layer files ([...]<layername>.clayer) as   well as copying associated files. One or more clayer files are required.
-- | -- | --

==> so it says "as well as copying the associated files".

When doing some attempts I am under the impression that associated files like RTE folder content and so on are created:
- either when opening the cprj file with Keil uVision (>=5.32)
- or when calling cbuildgen make after the compose stage

Is this correct ? 
Does "cbuildgen compose" only forge the cprj file or does it perform more actions than this ? (if so what is copied?)
Could you please possibly clarify ?

This behavior is fine, I'm just checking what is actually done by cbuildgen compose.

Thanks & regards,
Fred

jkrech, 2021-01-14:
Hi Fred,

for software components only the configuration files are located in the project directory under the RTE directory. If a software component is associated with a layer cbuildgen needs to retain these files when extracting into layers and composing from layers.

all files from the <files> section of the cprj file associated with a certain layer need to be retained relative to the cprj/clayer file.

The effect that you are seeing is that for component's configuration files both tools uVision and cbuildgen are copying the default version of configuration files to the corresponding directory under RTE if they are not found. Note that this will only have the desired outcome if your application uses an unmodified version of the configuration files. Usually the expectation for a ready to use/configured project is that all project specific files are shipped as part of the project. The RTE folder with the component's configuration files should be present.

Regards,
Joachim

fred-r, 2021-01-14:
Hi Joachim,

I am not sure to understand this statement:

"all files from the section of the cprj file associated with a certain layer need to be retained relative to the cprj/clayer file."

==> I do not see any artefacts being created in my file system when calling 'cbuildgen compose'.
How is this statement achieved ?

My current assumption is that I can :
1. call cbuildgen compose to obtain the cprj
2. then later on call cbuildgen make that will rely on the cprj to find associated files (no context related to cprj compose needed)

Basically the question behind this is: can I forge manually my cprj (without using cbuildgen compose) ? 

Thanks & Regards,
Fred

jkrech, 2021-01-14:
Hi Fred,
sorry,  I don't seem to be good at explaining it. I have updated the above comment.
The brief description of the compose command is probably misleading. The only thing the compose command actually does is to merge the specified layer files into a project.

a) you create a project (e.g. in uVision)
b) you define layers and assign files and components to a layer (e.g. in uVision)
c) export to cprj format
d) you run the cbuildgen extract, which will create individual folder and clayer files for each layer defined in your project
e) you run the cbuildgen compose, specifying the above clayer files and should recreate the same project as in c), unless you have left components or files unassigned. Those will be missing.
This by itself does not sound useful, but assuming that you are mixing and matching clayers extracted from various projects allows you to create projects from combinations.

Does that concept resonate with you?

Regards,

Joachim

fred-r, 2021-01-14:
Hi Joachim,

no problem with your explanations, sorry, I think I am a bit slow to understand the concepts and need to visualize things :-)

Now this is what I understand: 
1) the compose command is only creating a cprj by merging clayer files, nothing more.
2) when you open the cprj in Keil uVison >=5.32 then a project file is created listing the components
3) meanwhile, the tool resolves the components-file associations to populate the workspace

And when it comes to cbuildgen this step (3) is done at build time, not at compose time.

Is my understanding correct ?
(sorry again for spamming you with basic questions)

Thanks & Regards,
Fred

Then when we load the cprj with Keil uVision the files are listed and added to the project

jkrech, 2021-01-14:
Hi Fred,

1. correct, the cprj is created from the clayer files. In addition also the files specified in the clayer are copied from the layer folder to the project folder
2. correct, you can think of it as a conversion from cprj into uvprojx format. But the imported information is not limited to the selected components. It also adds the other files from the `<files>` section of the cprj to the project view. 
3. correct, uVision uses the loaded component description (*.pdsc) in order to identify and display the files in the project view. The build engine of uVision will use this information to compile, assemble and link the application.

Correct, cbuildgen also needs to identify the files that belong to each component listed in the cprj file. 

Regards,

Joachim

fred-r, 2021-01-14:
Hi Joachim,

thanks for your patience and explanations.
So now I think that I may very well skip the 'cbuildgen compose' stage by forging my own cprj (getting rid of clayers) and then call only 'cbuildgen make'.

Thanks & Regards,
Fred

jkrech, 2021-01-18:
Hi Fred,

sorry, you lost me on the question above. I guess if you don't use layers, you don't run the compose command and you don't copy a cprj and delete the other files.
As explained in another issue, the use of layers is fully optional and is independent from using packs and components.
Can we close this issue?

Regards,

Joachim

fred-r, 2021-01-18:
Hi Joachim,

that's clear now, thanks.

Thanks & Regards,
Fred

closed 2021-01-18 by fred-r