ARM-software/CMSIS_5 #1095: clayer versus CMSIS pack

fred-r, 2021-01-15:
Hi,

it is a bit unclear to me how the concepts of CMSIS pack and clayer work together.
A CMSIS pack is a way to define a re-usable component and it describes it with a pdsc.
A clayer is also a way to define a functional block (can be derived from a pdsc I guess) that will be used to create a cprj.

Doc states:
"Assuming that there is layers for multiple boards and applications sharing common interfaces between layers, any combination of HW laywer with an application layer can be combined to create projects and build consistent applications across a range of hardware."
=> so it seems that layers aimed at making easy to re-use components

But isn't CMSIS pack doing the same ?

The doc states:
"CMSIS-Pack describes a delivery mechanism for software components"
=> it insists on the delivery mechanism but to me it means you deliver a component this way.

I'm trying to understand the tole of the clayer files.
In my current view I do not understand why I can't go directly from pdsc to cprj.

Thanks & Regards,
Fred

jkrech, 2021-01-15:
Hi Fred,

did you already take a look here: [CB_Lab4Layer](https://github.com/MDK-Packs/CB_Lab4Layer#cb_lab4layer)
Here you can see layers at work.

There is no need to use layers at all. It is an optional feature, we have added to CMSIS-Build as an experiment.
You can use just plain cprj files to describe your project.

When creating projects you always need to add files to the project, select components and configure them to work for your project. Now assume you use the same basic board setup for multiple applications. What if you could divide your project into two layers (two partial, incomplete projects *.clayer). You just maintain the basic board setup layer as part of one project and then update it all application projects with it whenever you see a need for change. If you work with standard APIs between layers you can even share the application layers across multiple boards (see lab4layers).
This is a feature that relies on the cprj description format and the cbuildgen to extract, compose, add and remove layers. 
Note, it is not required that the projects use components from packs, but cprj requires the device descriptions from Device Family Packs. 

Regards,

Joachim

fred-r, 2021-01-18:
Hi Joachim,

thanks for the explanations.

Thanks & Regards,
Fred

closed 2021-01-18 by fred-r