ARM-software/CMSIS_5 #1097: how to use CMSIS with a given MPU

jerabaul29, 2021-01-18:
I am a bit confused about how to "install" CMSIS / how to use it with a given MCU.

For example, I am working on a project using the Apollo Ambiq 3 BLU, made available through the Sparkfun Artemis development board (https://www.sparkfun.com/products/15444), using the MBed-OS core made available by Sparkfun (https://github.com/sparkfun/Arduino_Apollo3).

What are the recommended steps / how should I "install" CMSIS to make it possible to use from my Arduino MBed-OS core sketches? Both in this specific case, and in general on other MCUs? I looked in the documentation (both readme and documentation website) but I could not find simple instructions for that - I am probably missing something, sorry.

This is related to:

https://github.com/sparkfun/Arduino_Apollo3/issues/326

https://github.com/sparkfun/Arduino_Apollo3/issues/325

JonatanAntoni, 2021-01-18:
Hi @jerabaul29,

I regret to say, your question is too unspecific to receive a specific answer.

The official way of "installing" CMSIS is by installing the released CMSIS-Pack. This requires a CMSIS-Pack compliant development environment (such as MDK, ArmDS or IAR Embedded Workbench).

mbedOS is based upon CMSIS (e.g. using CMSIS-Core and CMSIS-RTOS components), already. Adding further components using the mbed development environment (e.g. mbed cli) is a matter of mbed, not of CMSIS. Please consider consulting the mbed documentation and community.

If you have specific questions about CMSIS, e.g. about one of its components, I am happy to help.

Cheers,
Jonatan

jerabaul29, 2021-01-18:
Ok, I perfectly understand, many thanks for your help!

Actually this is already very helpful: if I understand well, what you mean is that I should ask MBed-OS for help here, right? :) It is already a good start to know where I should discuss this :) .

Feel free to close :) (maybe once we can confirm where I should ask for the integration to be put in place :) ).

jerabaul29, 2021-01-18:
To be a bit more specific: what I want is to be able to use the DSP part of this repo, in particular FFTs.

JonatanAntoni, 2021-01-18:
Hi @jerabaul29,

yes, I guess your question is mbed related. From CMSIS point of view, you just need to add the source files to your project and have the header in your include path. That should be enough to use DSP functionality.

Cheers,
Jonatan

PS: Typically, I leave it up to the requester (you in this case) to close the issue once the answer is satisfying.

jerabaul29, 2021-01-18:
Ok, many thanks for your help! 

The Mbed-OS question was closed quite fast, I do not think they want to really integrate this in their codebase. Their answer (copied from the issue referenced a few lines above):

"
It should work to download dsp, add it to an application, configure it (I would assume it requires some symbols to be present, they either come from cmsis/mbed files or an application - I would start there to check if the configuration is done).
"

The only thing that slows me down a bit here is that it looks like this means that I would need to:

- put the fft implementation in the right folder: https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_f32.c

- put the headers imported by that implementation in the same folder, possibly modifying pathes I suppose:

https://github.com/ARM-software/CMSIS_5/blob/f9d4b2adad74bc44aaaf4c705594f3550edd0ad1/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_f32.c#L29

- this header https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Include/dsp/transform_functions.h  in turn seems to import more headers:

https://github.com/ARM-software/CMSIS_5/blob/f9d4b2adad74bc44aaaf4c705594f3550edd0ad1/CMSIS/DSP/Include/dsp/transform_functions.h#L29-L36

which again import more headers etc.

So I suppose this means I should somehow in the end either do a huge manual work or, much better, have the full CMSIS DSP folder as a git submodule then, or something like this (?). Not sure how this would work with the content of some separate files such as these located there: https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP/Include while the sources I try to compile are here: https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP/Source/TransformFunctions . I suppose somehow this building logics is built in the different build files available there, not sure if there is a way to use them as-is without the need to do some more work.

Another thing is that I would be interested in having a bit general solution; I can hack my way into getting a rfft_float32 function to compile, but I suppose other users will want other parts of the DSP for their applications.

Probably just a n00b "howto sofware build" question.

jerabaul29, 2021-05-03:
Sparkfun was helpful and did the work of makind CMSIS available on their arduino core :) . Closing.

closed 2021-05-03 by jerabaul29