ARM-software/CMSIS_5 #1101: wrong clock phase in DP_SW.c

nerdralph, 2021-01-21:
SWDCLK idle state is low, so a clock cycle should go high, then back to low.
https://github.com/ARMmbed/DAPLink/issues/764

RobertRostohar, 2021-01-22:
**Copying the reply from ARMmbed/DAPLink#764**

> The spec says: "The target writes data to SWDIO on the rising edge of SWDCLK. The target reads data from SWDIO on the rising edge of SWDCLK."

I guess you were looking at the specification from Arm (example: DSTREAM debug probe specification):
[DSTREAM SWD timing requirements](https://developer.arm.com/documentation/dui0499/d/ARM-DSTREAM-Target-Interface-Connections/SWD-timing-requirements?lang=en)

> Although the ARM SWD spec says, "The probe reads data from SWDIO on the rising edge of SWDCLK."

Exactly. This is also how the SW_READ_BIT macro is implemented.

The macro for reading a bit in SW_DP.c:
`#define SW_READ_BIT(bit) \`
`  PIN_SWCLK_CLR(); \`
`  PIN_DELAY(); \`
`  bit = PIN_SWDIO_IN();  \`
`  PIN_SWCLK_SET(); \`
`  PIN_DELAY()`

> SWD_READ_BIT should be reading the SWDIO line after setting SWCLK, not before.

Not really. The target will start driving the next bit on the rising edge and sampling after the rising edge might read the next bit. 

Keep in mind that the macro execution can be interrupted (ISR) which would make the time between sampling and rising edge even longer.

> SWCLK idle state is low

It is not really about SWCLK idle state but rather at which edge the data is driven and sampled. However the idle state would normally be high. You can see this also from the timing diagram in the previous mentioned specification.

> I found a better [description](https://www.cnblogs.com/shangdawei/p/4757044.html) that says the probe should read after the falling edge.

I'm not sure about 'better'. Arm defined the specification so I would stick to the official Arm specification.

Regarding the approach that probe should read the data on the falling edge as mentioned in the quoted description:
That is a possibility. It will however decrease (half) the maximum clock. The maximum clock can be achieved by sampling later (rising edge). Of course it is assumed that both SWCLK and SWDIO have similar electrical characteristic.

BTW: That same description also shows waveforms with SWCLK idle state being high.

There are a lot of CMSIS-DAP based debuggers out that are used with many different targets and there are no real problems due to the discussed sampling approach.

Also professional debuggers as DSTREAM, ULINKpro and ULINKplus.

nerdralph, 2021-01-22:
I corrected my comment about SWD_READ_BIT: "Therefore I think SWD_READ_BIT should sample SWDIO, set SWDCLK high, and then set SWDCLK low again."  We both agree that SWDIO must be sampled before SWDCLK goes high.

I agree that, despite the spec stating the probe reads data on the falling edge of SWCLK, reading just before SWCLK high might permit higher speeds on targets with very low slew rates.  However with 3V3 IO rise times of 3-5ns, that's not a real problem, and I think it's best if the code, "stick to the official Arm specification".

I don't agree that idle state is high.  Every ARM target I've tested (STM, GD, CHK) pulls SWCLK low and SWDIO high when SWD is enabled.  That's the idle state.  In addition to the Segger J-link implementing it that way, I've also looked at the ST-link built into the nucleo boards, and it does the same.

RobertRostohar, 2021-01-22:
> "stick to the official Arm specification"

Which CMSIS-DAP does: "The probe reads data from SWDIO on the rising edge of SWCLK."

SWCLK idle state is less relevant and it should work with both high or low idle state. Some debug probes use high state (CMSIS-DAP based probes, ULINK family, DSTREAM) and some use low (J-link, ST-link).

If the target pulls SWCLK high or low depends on the actual target device and any external resistors.

For example some devices leave the SWCLK input floating (ex: NXP LPC43/18) and rely on external resistors, some devices have fixed internal pull-down or pull-up resistors and some devices have configurable internal pull-down or pull-up resistors.

External resistors seem to be more often pull-down resistors but also pull-up resistors are being used.

For example DSTREAM shows a typical [SWD connection](https://developer.arm.com/documentation/dui0499/d/ARM-DSTREAM-Target-Interface-Connections/SWD-connections?lang=en) with SWCLK pulled up.

Others might suggest SWCLK pulled low as for example this article from NXP [Design Considerations for Debug](https://community.nxp.com/t5/LPCXpresso-IDE-FAQs/Design-Considerations-for-Debug/m-p/469565). But it also states: "TCK/SWCLK, although a pull-down is recommended, you can alternatively use a pull-up. The main thing is that it does not float."

To summarize: 
This issue which is named: "wrong clock phase in DP_SW.c" doesn't seem to be an issue.

nerdralph, 2021-01-22:
Pulling SWCLK high for idle instead of low means a spurious bit being received by the target before communication starts.

I've double and triple checked the spec, and don't see anything that confirms your position (the code in DP_SW.c) that a clock cycle starts low and ends high.  Figure 4-1 and 4-2 both show the first phase of a clock cycle high, and the 2nd phase low.
https://static.docs.arm.com/ihi0031/c/IHI0031C_debug_interface_as.pdf

The SW_ macros have the the clock phase inverted.

nerdralph, 2021-01-23:
Here's Figure 4-1 showing a clock cycle starts with the high phase and end ends low:
[image]

RobertRostohar, 2021-01-25:
> Pulling SWCLK high for idle instead of low means a spurious bit being received by the target before communication starts.

This can occur only when debug probe starts driving SWD lines and assuming that SWDCLK has a pull-down resistors (might not be the case – pull-up resistor might be used). However the debugger will always issue a reset sequence (see chapter 4.4.3 in the mentioned document) ensuring that there are no unintentional transfers due to hot-plugging.

The mentioned document clearly states the following in chapter 4.1: “This chapter does not describe the physical characteristics of the SWD interface, such as signal timings.” 

Also directly before Figure 4-1 there is a note: “The diagrams in this section are included to show the operation of the Serial Wire Debug protocol. They are not timing diagrams for the protocol.”

Therefore this chapter is not relevant.

Chapter 4.4 gives and overview of the physical Serial Wire Debug interface.

[image]

The figure 4-8 shows that a bit starts with SWDCLK being high, falling edge followed by low half period, rising edge followed by high half period. This looks as high idle clock.

Even on the page that you have quoted as a better [description ](https://www.cnblogs.com/shangdawei/p/4757044.html) a high idle clock is shown.

[image]

As initially stated. It should not really matter if idle clock is high or low. Relevant is when the probe and target drive the lines and on which edge they sample the data.

nerdralph, 2021-01-25:
@RobertRostohar
Why are you introducing straw man arguments about signal timings?  Where did I say anything about signal timing?
Since it appears you work for ARM, I'm sure you know what timing diagrams are.

I'll admit s. 4.4 (which wasn't in the original 0031A doc), shows the clock phase as it is implemented in DP_SW.c.  Therefore I've changed the issue title from "wrong clock phase" to "preferred clock phase".

I seem to be wasting my time trying to convince you that the best implementation is with SWCLK idle low, and the first phase of the cycle with SWCLK high.  As I've already mentioned, the ST-Link and J-Link chose to implement it that way.  So if the code isn't going to be changed, I think there at least should be comments or documentation added making it clear that the CMSIS-DAP implementation chose to have the clock polarity low when idle.

RobertRostohar, 2021-01-25:
> Why are you introducing straw man arguments about signal timings? Where did I say anything about signal timing?

I was just trying to highlight that the diagram in Figure 4-1 should not really be a reference for the clock phase. The purpose of those figures is to explain the protocol on a higher level (order of bits and their meaning but not looking inside a bit). At least this is how I understood the notes in the document.

> the best implementation is with SWCLK idle low, and the first phase of the cycle with SWCLK high

I still don't see the benefit of having SWDCLK idle low and the impact if it is high.