ARM-software/CMSIS_5 #1103: Can we have amore generic RTE_Components.h ?

fred-r, 2021-01-25:
Hi,

the CMSIS pack spec provides this feature: "Sometimes, software components need to know of other components. Depending on the availability or configuration of another component, certain settings may be required. A distinct header file is available to make components aware of each other: RTE_Components.h."

This feature is nice but all compiler switches end up in the very same file.
Would it be possible to specify in which file the definition should go ?
This would be better from a modularity perspective (each layer could have its file listing the selected components).

Thanks & Regards,
Fred

jkrech, 2021-01-25:
Hi Fred,
could you please clarify whether the term "layer" above is referring to a CMSIS-Build layer?
Where would the filename be specified? Is this part of the layer description?
Maybe you are looking for an additional file rather than a replacement for RTE_Components.h ?
I have not got exactly what you try to achieve, especially since layers is an optional concept.

Regards,

Joachim

JonatanAntoni, 2021-01-25:
Hi @fred-r,

The `RTE_Components.h` file is a central configuration file, intentionally. Of course this file should *not*
 be used for component specific configuration (that might interfer with other components). In addition components can contribute to either a `Pre_Include_Global.h` or a `Pre_Include_<Cclass>_<component>.h` file.

Can you please give me some more details about a concrete use case you are after?

Cheers,
Jonatan

fred-r, 2021-01-25:
Hi,

basically the target layer of my project (ending up in a cprj file) is made of several optional components.
What I would like is to get a header file which is not global to the project to specify the components activated in this layer.
I may have the same need at application layer too.
But of course we can decorrelate this from the layers concept.
Basically, I am looking for a way to eable switches in specific header files, not only in one global file.
I understand the need for the global file and we can keep it but I would also need a finer-grain granularity allowing me to indicate how my software is composed.

Thanks & Regards,
Fred

jkrech, 2021-01-26:
Hi Fred,
I don't see why we would duplicate this information on a "local" as well as "global" level yet.
We can add more code generation capabilities to the component description but I wonder where we draw the line to using a generator component to handle this type of inter component dependencies?

Regards,
Joachim

fred-r, 2021-01-29:
Hi,

so can this proposal be taken into account ?

Thanks & Regards,
Fred

jkrech, 2021-02-04:
Hi Fred,

I am seriously concerned about the usability of your proposed feature as for each project you have the freedom to specify one or more header file names as well as the selection of components which shall be reflected in that header file and you also need to add the reference to these headerfiles within certain modules which must not be read-only as you specify the filename within the project. 
I seem to be missing something here... but this does not sound like an easy to use, scalable approach to me.

Regards,

Joachim

Regards,

Joachim