ARM-software/CMSIS_5 #1104: cbuildgen : /cprj/files/file are ignored

slhultgren, 2021-01-25:
Group-less files are ignored by cbuildgen:

```
<files>
  <file category="sourceC" name="./test.c" />
</files>
```
test.c will not be built or even output into the generated makefiles using:
```bash
$ rm -rf build_test && cbuildgen test.cprj make --toolchain=GCC --outdir=build_test --intdir=build_test && make -f test.mak -C build_test
```

Using cbuildgen on linux: 
(cbuildgen): Build Process Manager 0.10.0 (C) 2020 ARM

brondani, 2021-01-25:
Hi, thanks for reporting this!
As in the mentioned issue, files outside groups are not yet supported.
For the time being please use the normal structure also used in the examples:
```
<files>
  <group ... >
    <file ... >
    </file>
  </group>
</files>
```
Thanks
Daniel

brondani, 2021-09-23:
Hi @slhultgren,
Now CMSIS-Build supports multiple groups as well as group-less files.
Can you please check the [CMSIS-Build Version 0.10.3](https://github.com/Open-CMSIS-Pack/devtools/releases/tag/tools%2Fbuildmgr%2F0.10.3) release in the [Open-CMSIS-Pack](https://github.com/Open-CMSIS-Pack/devtools) repository and confirm whether it meets your expectations?
Thanks
Daniel

slhultgren, 2021-09-28:
Hi @brondani, I confirm that this now works well in 0.10.3. Thanks for the update! :)

closed 2021-09-28 by slhultgren