ARM-software/CMSIS_5 #1105: cbuildgen : exit code success even though error during generation

slhultgren, 2021-01-25:
if the cprj file contains a file that is missing:
```xml
<files>
  <group name="test">
    <file category="sourceC" name="./missing_file.c" />
  </group>
</files>
```
cbuildgen will output an error, but the exit status is still 0 on Linux which makes scripting and error detection unreliable:
```bash
$ rm -rf build_test && cbuildgen big_test.cprj make --toolchain=GCC --outdir=build_test --intdir=build_test && echo "cbuildgen exited with $?"
(cbuildgen): Build Process Manager 0.10.0 (C) 2020 ARM

*** ERROR M204:
  Path not found: '/local/home/hultgres/dev/cube2.0/infra/clone2/infra/Firmware/samples/HAL/CRC/CRC_UserDefinedPolynomial/gen/file_10e0.c'!

M652: Generated makefile for project build: '/local/home/hultgres/dev/cube2.0/infra/clone2/infra/Firmware/samples/HAL/CRC/CRC_UserDefinedPolynomial/build_test/big_test.mak'
cbuildgen exited with 0
```

Using cbuildgen on linux:
(cbuildgen): Build Process Manager 0.10.0 (C) 2020 ARM

brondani, 2021-01-25:
Hi, thanks for reporting this!
It's a bug and will be fixed for the next release.
Thanks
Daniel

slhultgren, 2021-09-28:
Hi @brondani, I confirm that this now works well in 0.10.3 for missing files. Thanks for the update! :)

closed 2021-09-28 by slhultgren