ARM-software/CMSIS_5 #1111: Incorrect DSP inline macro definitions in arm_math_types.h

stephanosio, 2021-01-27:
As far as I can see, `__attribute__((inline))` is not a valid attribute for GCC.

https://github.com/ARM-software/CMSIS_5/blob/a614de942bcee6df533ea1231898872fe06217f5/CMSIS/DSP/Include/arm_math_types.h#L67-L73

This should be fixed as follows:
```
#define __STATIC_FORCEINLINE static inline __attribute__((always_inline)) 
#define __STATIC_INLINE static inline
```

christophe0606, 2021-01-27:
@stephanosio I'll correct it.
But the code you quoted is only used when compiling the Python wrapper on host.
So, you should not see the problem if the CMSIS compiler headers are used since they use a correct gcc definition.

Unless of course if your are compiling the Python wrapper ...

stephanosio, 2021-01-27:
> But the code you quoted is only used when compiling the Python wrapper on host.
> So, you should not see the problem if the CMSIS compiler headers are used since they use a correct gcc definition.

That is correct. We've encountered this issue in [a Zephyr PR](https://github.com/zephyrproject-rtos/zephyr/pull/31463) which attempts to build CMSIS-DSP for the Zephyr native POSIX arch (i.e. host) using `__GNUC_PYTHON__`.

christophe0606, 2021-01-28:
Should be correct in https://github.com/ARM-software/CMSIS_5/commit/71627bc91534ed9eec2361c0ef6442cd057653e0

stephanosio, 2021-01-29:
Thanks for looking into this. Closing as the issue has been resolved.

closed 2021-01-29 by stephanosio