ARM-software/CMSIS_5 #1112: File 'startup_stm32f769xx.c' source must have 'attr="config"'

fred-r, 2021-01-27:
Hi,

we are working on the creation of our own DFP packs and we face some warnings when creating the packs since we have removed the 'attr=config' attribute for the startup files:

Warning detected in DFP packs:

Phase3: RTE Model based Data & Dependencies check

*** INFO M377: STMicroelectronics.STM32F7xx_DFP.pdsc (Line 473)
  File 'startup_stm32f769xx.c' source must have 'attr="config"'

The pdsc is like this:
        <file category="source" condition="STM32F769_ARMCC" name="Source/startup_stm32f769xx.c" version="0.0.6"/>

instead of initially:
         <file category="source" condition="STM32F769_ARMCC" name="Source/startup_stm32f769xx.c" attr="config" version="0.0.4"/>

I did not see anything in the spec stating that startup file must be configured as 'config' file.
Why do we have this warning ?
What are the implications ?

Thanks & Regards,
Fred

JonatanAntoni, 2021-01-27:
Hi @fred-r,

This rule is not due to a limitation in the spec of CMSIS-Pack but its how the CMSIS startup is intended to be used.

A user might need to modify the startup file (either .c or .s), e.g. because some custom code is required to be added to the Reset_Handler *or* modifications to the IRQ table are required.

If you are absolutely sure that users of your DFP should not modify the startup code, you may disable that check when running PackChk with `-x M377`.

Please keep in mind that such a change might not be backward compatible. Without attr="config" the version tag is not required as well.

Cheers,
Jonatan

ilg-ul, 2021-01-27:
> we are working on the creation of our own DFP packs ... File 'startup_stm32f769xx.c' source

If you are working on your own packs, could you not repeat the usual mistake of bundling together the startup code with the vectors?

In my projects I have a single `startup.c`, which I use for all projects, and multiple `vectors_<device>.c` files, with the vectors.

If, for any reasons, you insist on packing your startup with the array of pointers to interrupt handlers, please consider a mechanism to disable your startup code and replace it with user provided code.

JonatanAntoni, 2021-01-27:
Breaking the `startup.c` apart and put the vercor table into a separate file can be discussed. Typically both files are somehow device specific.

Diverging from the CMSIS templates without enhancing these as well looks very odd to me. Having some sort of similarity in these common project parts is the major goal of CMSIS. This shall make it easier for users coping with different Arm-based devices, e.g. interrupt tables are always found in the same place. Hence I'd like to ask you *not* to diverge but rather collaborate on enhancing the templates.

ilg-ul, 2021-01-27:
> Typically both files are somehow device specific.

Nope. The content of the array of vectors is device specific. The startup is **application** specific. It becomes device specific by calling various device specific initialisation routines. 

Bundling the vectors with the startup code, and requiring them to be in an assembly file was one of the initial CMSIS design mistakes, and now the majority of the CMSIS Packs use assembly startups, making them useless for my development environment.

Now I saw that the `startup_Device.c` is no longer an assembly file, which is good, and it no longer bundles the startup code, which is also good, but the content of `Reset_Handler()` is fixed and there is no way to disable it. 

I suggest you make `Reset_Handler()` weak, or use any mechanism with a similar effect (for example a macro `USE_CUSTOM_RESET_HANDLER` which can be passed on the command line, and conditional preprocessing; etc).

> Diverging from the CMSIS templates without enhancing these as well looks very odd to me. 

The suggestion to define the vectors in C and to make the startup optional was one of the many suggestions I made to the initial CMSIS design (see the many issues I registered to the initial CMSIS project, before CMSIS_5), and we also discussed it during my visit to Bayern (many years ago); the general understanding was that something must be done, but unfortunately no actions were taken. :-(

JonatanAntoni, 2021-01-27:
> Now I saw that the startup_Device.c is no longer an assembly file

This on its own has issues.

E.g. you need to take care not to compile this file with hardware floating point mode. If you do it might happen that the compiler chooses to make use of co-processor instruction *before* the co-processor interface is enabled by the startup code itself. Crash.

Or you might end up with using the stack in an inappropriate way. Compiler and hardware engineers still recommend to use assembly language in order to keep full control over the startup code.

> I suggest you make Reset_Handler() weak

The startup.c file is intended to be a user code template. It shall be copied into the project to be modified by the user. Same holds true for the vector table. While the default should be fine for most user there might be use cases where one needs to tweak it. That's why it is marked with `attr="config"` in the packs.

ilg-ul, 2021-01-27:
> E.g. you need to take care not to compile this file with hardware floating point mode. If you do it might happen that the compiler chooses to make use of co-processor instruction before the [...]

Right, things can go even weirder, modern compilers may consider some low level code as undefined behaviour, and generate unexpected results.

But this can be easily overcome, you need to write only the initial part of the `Reset_Handler` in assembly, which can be done with `asm()` in a C file, take care of the very early inits, then call a `_startup()` routine written in C.

But, as I said, all those details are application/framework specific, the CMSIS provided code should simply not stand in its way, as it does now.

> The startup.c file is intended to be a user code template. It shall be copied into the project to be modified by the user. 

If you comment out the `Reset_Handler`, and add a comment explaining that the user should provide the content in the application, the file itself can be used without any changes, greatly simplifying project creation.

ilg-ul, 2021-01-27:
FYI, here is how a `vectors_<device>.c` file looks like in my framework:

- https://github.com/micro-os-plus/devices-stm32f4-xpack/blob/xpack-develop/src/vectors/vectors_stm32f407xx.c

It is basically what you have in your template, except the `Reset_Handler`, which is in another file.

Since the CMSIS Packs do not provide such files, I have to generate them for each device family, which adds some maintenance time.

JonatanAntoni, 2021-06-08:
@fred-r, is this issue solved or is something open to be discussed, here? If not, may I ask you to close the issue, please? Thanks.

fred-r, 2021-06-17:
I assume using -x M377. is fine, thanks.

closed 2021-06-17 by fred-r