ARM-software/CMSIS_5 #1113: Question about file version

fred-r, 2021-01-29:
Hi,

the CMSIS documentation states:

File-specific version information. This is used particularly for files copied into the project workspace. Before a file gets copied, a version check avoids unnecessary copy actions. If a file does not have a version, then the component version is used. The version format is described in Version Type.

I want to make sure I understand properly : is the file version used for files which are not copied in the workspace ?
Is it possible to say that I want a condition depending on a specific version of a file ?
Today this looks limited to components version.

Also, regarding the files copied in the workspace.
Do you mean that the tool compares the version stored locally versus the version available in the pack ?

Thanks & regards,
Fred

jkrech, 2021-01-29:
a) Is the file version used for files which are not copied in the workspace ?
No, the version attribute of the 'file' tag is ignored unless the attr="config" is set.
b) Is it possible to say that I want a condition depending on a specific version of a file ?
No, conditions can only specify a component and not a file within a component. I cannot see how this would be useful.
c) Do you mean that the tool compares the version stored locally versus the version available in the pack ?
Yes.
One of the major benefits of using CMSIS-Pack is the fact that I can specify a project without "fixing" the versions of the components I want to use. If I install a different version of a pack that contains my components, the component versions are likely to be different as well. The build generator will locate the component modules in the new pack to build the project.  However config files have been copied into the project and are probably edited to fit the application.
The version of the config file is stored as part of the component selection in the cprj file at the point when the file got copied into the project. If the new version of a component has a newer version of the config file the semantic version comparison will tell you whether merging of the new config file from the pack with the modified config file in your project is required.
In MDK the component icons will indicate whether a patch, minor or major version difference exists. You can launch a merge tool from the IDE to migrate your configuration file.
In cbuildgen there is a command line option to configure a merge tool which will attempt to merge compatible config files or flags an error if not successful.

Regards,

Joachim

fred-r, 2021-01-29:
Thanks Joachim, very clear.

JonatanAntoni, 2021-06-08:
Closing due to inactivity. Please feel free to reopen if anything left to be discussed.

closed 2021-06-08 by JonatanAntoni