ARM-software/CMSIS_5 #1115: SVDConv universal prefix on peripheral derived from other peripherals

conkerkh, 2021-01-31:
Suppose we have two UARTS: UART0 and UART1, UART1 will derive from UART0.

When SVD Conv generates header files, peripherals that derive from others will all have a prefix named after first peripheral; something like UART0_TypeDef, macros will start with UART0 etc. It's quite strange to have UART0_Type for UART1. I think that SVD Conv could detect it has multiple derivatives of one peripheral, in many cases they will be numbered as UART0, UART1, UART2 and so on. Simply using UARTx as a prefix would be much more readable and logical here.

jkrech, 2021-02-01:
Hello,
have you tried specifying a <headerStructName>
see:
https://arm-software.github.io/CMSIS_5/SVD/html/elem_peripherals.html#elem_peripheral
Regards,
Joachim

conkerkh, 2021-02-02:
I checked it and I think headerStructName mostly does what it should. However macros will still have peripheral prefix instead of using headerStructName, I belive the behaviour should be either to use the same headerStructName with macros, or define headerMacroName.