ARM-software/CMSIS_5 #1123: Suffixed macros with latest update of SVDconv

sebastienmasson, 2021-02-10:
We have been using SVDconv V3.3.21 until now.
We have recently updated to SVDconv V3.3.35.
Both running on linux.

We use the same command line: 
`./SVDConv <myFile>.svd --generate=header --fields=macro --fields=struct --strict -o <output path>`

We observe the followings:
- With the previous version (3.3.21): Unique peripherals and multi-instanciated peripherals sharing the same values were all sharing the same macro names:
- With the latest version (3.3.35): All peripherals are suffixed with an instance number (starting with 0).  And, multi-instanciated peripherals sharing the same values are still present only once.

This as a double consequence:
- Existing drivers stopped working;
- We are forced to address peripheral XXX2 using macros made for peripheral XXX0.

A typical example are the GPIOs (but we observed the same issue with many peripherals)
- Former version: `#define GPIO_CTRL_RST_Pos`
- Latest version: `#define GPIOA_CTRL_RST_Pos`

My assumption is this related to the parameter "derivedFrom" in the tag "Peripheral" within SVD:
`<peripheral derivedFrom="GPIOA">`

I cannot find out a way in the documentation to keep the same behaviour as before.
(From here: https://www.keil.com/pack/doc/CMSIS/SVD/html/svd_SVDConv_pg.html)

It's fine to me if we have new macros but I would definitely recommand to keep the former names as well.
Is there any command line argument to force backward compatibility?

I thank you for you help.
KR
Seb

sebastienmasson, 2021-03-16:
This issue is still open and we have no fall back at the moment.  Do we have to manage the SVD otherwise?  It there a command line argument to pass to SVDconv?

This issue looks close to #1115 

Thank you

JonatanAntoni, 2021-03-17:
Hi @sebastienmasson,

thanks for reaching out and sorry for the delay.
We have conflicting requirements in SVDConv here. Some prefer it the new way, other prefer the former way. 

It turned out that the former way causes issues in certain situations. Hence we needed to adopt. Sorry, if this is now causing inconvinience for you.

Cheers,
Jonatan

sebastienmasson, 2021-03-17:
Hi Jonatan,

Thank you for your reply.

The major issue the new way introduces is we have no backward compatibility.  It would be fine for me if we had command line options to ask for the old way, the new one, or both.

KR
Seb

sebastienmasson, 2021-09-24:
What is the status of this issue?  Any update?