ARM-software/CMSIS_5 #1126: cachel1_armv7.h : loses information [clang-diagnostic-error]

lantiati, 2021-02-16:
Hi,
We are running clang-tidy check our code, and the following errors are generated :

\CMSIS\Include\cachel1_armv7.h: error: cast from pointer to smaller type 'uint32_t' (aka 'unsigned int') loses information [clang-diagnostic-error]
uint32_t op_addr = (uint32_t)addr
^

This error aborts the clang-tidy check.

If I modify the code like below, the compilation is ok:
uint32_t op_addr = * addr;

Thank you for the help!
Attila Lantos

JonatanAntoni, 2021-02-16:
Hi @lantiati,

Agreed, such low level code often causes warnings in static analysis. Please be aware that the code change you did changes the code behavior! Replacing a cast with a pointer-dereference is not equivalent!

The warning might be caused by (and is correct only if) assuming `sizeof(void*) == 64`. CMSIS as it stands today is 32-bit only (which Armv7-M is), hence `sizeof(void*) == 32`. In this case an `uint32_t` variable is large enough to hold the address value. There is no information loss, here.

Cheers,
Jonatan

rsaxvc, 2021-03-04:
Would uintptr_t be a better and more clear variable type here?

JonatanAntoni, 2021-03-04:
Hi @rsaxvc,

Good catch. We need to check if `uintptr_t` is defined properly for all target compilers. I am happy to receive contribution on this one.

@lantiati, would this solve your diagnostic warning as well?

Thanks,
Jonatan

rsaxvc, 2021-03-08:
It's technically an optional C99 feature, but it's been supported for quite some time by many compilers / c-libraries targeting ARM cores, including DS5(I'd have to check but I think RVCT3.x and RVCT4.x did as well), clang. IAR7.50 and newer has it(but older perhaps too, I'm not too familiar with ). For GCC I think it would depend on the C library used, but newlib has had it since before 2010.

JonatanAntoni, 2021-03-08:
@rsaxvc,

Thanks for your comprehensiv analysis. I think if we can be sure that compilers used in active development as by today we are fine. I think we don't need to care about GCC versions more than 10 years old.

@lantiati, is it feasible for you to provide a patch (PR) if the proposed fix works for you? Would be great to get this contributed.

Thanks,
Jonatan

JonatanAntoni, 2021-06-08:
Hi all,

I've just taken a closer look into the code and to me it looks like just changing the data type used for `op_addr` doesn't solve the issue. The value is later written to the 32-bit register `SCB->ICIMVAU` which is defined `uint32_t` as well. Assigning a `uintptr_t` to a `uint32_t` register may cause the same warning. 

I'd like to ask you to carefully check the details and provide a proper soluion that works for you, please.

Thanks,
Jonatan