ARM-software/CMSIS_5 #1130: Error when building DSP library

Gostas, 2021-02-18:
I am trying to build the DSP library for the cortex-m4 cpu. The compiler I am using is arm-none-eabi-gcc 9.2.1 (9-2019-q4-0ubuntu1). After running make VERBOSE=1, I am getting the following error:

`/home/kpaps/Documents/CMSIS_5/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_init_f16.c:70:3: error: unknown type name 'arm_cfft_radix4_instance_f16'
   70 |   arm_cfft_radix4_instance_f16 * S,`

christophe0606, 2021-02-19:
@Gostas f16 is not always well supported depending on the core / compiler so there is some logic to enable / disable it.

But it looks like this logic is missing in arm_cfft_radix4_init_f16.c.

There should be a test on ARM_FLOAT16_SUPPORTED with #if/#else to disable or enable the compilation.

This flag is disabled in your compilation so the instance structure and tables for this f16 algorithm are not compiled but the function is compiled although it should not.

Note that:

- radix4 is deprecated. But we understand people are still using it ;
- If you don't need f16 (and it is only useful when there is some HW acceleration for f16 which is not the case on M4) you may remove the f16 files from the build

Gostas, 2021-02-19:
Thanks for the explanation @christophe0606 . I added the guard condition in arm_cfft_radix4_init_f16.c and was able to get it to build, because I wasn't sure how to remove the f16 files from the build.

christophe0606, 2021-02-19:
@Gostas I have just pushed https://github.com/ARM-software/CMSIS_5/commit/9f524153122d10a42d5e770904e5a488471fdd4d which should solve the issue.

Gostas, 2021-02-19:
Great, thanks!

closed 2021-02-19 by Gostas