ARM-software/CMSIS_5 #1135: RTX5 tracking of PendSV bit

tobermory, 2021-02-24:
I note that in RTOS2/RTX5, both os_systick.c and rtx_kernel.c have a boolean that tracks the state of the PendSV bit in the SCB->ICSR register. I am puzzled at this apparent duplication of effort. I imagine it would be a bad situation if one were false while the other true? 

Is the duplication down to the possibility of os_systick.c being usable by RTOS2-compliant kernels other than RTX?

RobertRostohar, 2021-02-24:
> I note that in RTOS2/RTX5, both os_systick.c and rtx_kernel.c have a boolean that tracks the state of the PendSV bit in the SCB->ICSR register. 

Not really. rtx_kernel.c tracks Pend**SV** (PendSV exception) where os_systick.c tracks Pend**ST** (SysTick exception).

tobermory, 2021-02-24:
Ah, I see my mistake now, thank you for the clarification.

JonatanAntoni, 2021-02-24:
@tobermory, please consider closing this issue if the anwer is satisfaying. Thanks.

closed 2021-02-24 by tobermory