ARM-software/CMSIS_5 #1138: building the DSP for use on a specific MCU

jerabaul29, 2021-03-03:
I want to be able to use some DSP functions on a specific MCU (in my case, Apollo Ambiq 3, but I guess the details are not important).

- From what I understand, this means that I need to build the DSP module for this specific microcontroller, and then later on to link to the built libraries during compilation of the program to be run on the MCU that uses some DSP functions, is that right?

- I then try to follow the Readme at https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP , section ```Building only the CMSIS-DSP library```, as all I want is to compile the DSP as a library for later use, is that right?

- I encounter a few difficulties there. Following the instructions, I have some issues if I use in the cmake function:

```-DCMAKE_TOOLCHAIN_FILE="../../CMSIS_ROOT/CMSIS/DSP/armac6.cmake"```

as is indicated in the Readme. If I do that, I get an error message:

```
CMake Error at /usr/share/cmake-3.16/Modules/CMakeDetermineSystem.cmake:99 (message):
  Could not find toolchain file: ../../CMSIS_ROOT/CMSIS/DSP/armac6.cmake
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)
```

I suppose that this indicates that the paths are wrong. Using instead path: ```-DCMAKE_TOOLCHAIN_FILE="../../armac6.cmake``` works fine though. Is this the correct approach / should the readme be updated?

- However, even fixing this, I still hit some error messages that I have no idea how to fix. Any hint?

```
~/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build$ cmake -DROOT="/home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS"       -DCMAKE_PREFIX_PATH="/home/jrmet/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc"       -DCMAKE_TOOLCHAIN_FILE="../../armac6.cmake"       -DARM_CPU="cortex-m4"       -G "Unix Makefiles" ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_b0f7c/fast && /usr/bin/make -f CMakeFiles/cmTC_b0f7c.dir/build.make CMakeFiles/cmTC_b0f7c.dir/build
    make[1]: Entering directory '/home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_b0f7c.dir/testCCompiler.c.o
    /usr/bin/cc   -mcpu=cortex-m4 --target=arm-arm-none-eabi    -o CMakeFiles/cmTC_b0f7c.dir/testCCompiler.c.o   -c /home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeTmp/testCCompiler.c
    cc: warning: ‘-mcpu=’ is deprecated; use ‘-mtune=’ or ‘-march=’ instead
    cc: error: unrecognized command line option ‘--target=arm-arm-none-eabi’
    make[1]: *** [CMakeFiles/cmTC_b0f7c.dir/build.make:66: CMakeFiles/cmTC_b0f7c.dir/testCCompiler.c.o] Error 1
    make[1]: Leaving directory '/home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeTmp'
    make: *** [Makefile:121: cmTC_b0f7c/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "/home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeOutput.log".
See also "/home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeError.log".
```

christophe0606, 2021-03-04:
What is your cmake version ? There are problems with recent versions and I never had the time to check this. I am using 3.14.3

When you made your first try, it created some folders and files in your build folder like CMakeFiles, CMakeCache.txt. Remove them. It may have cached some information from the first failed attempt.

Also you could try with an absolute path.

christophe0606, 2021-03-04:
@jerabaul29 Something like that:

cmake -DCMAKE_PREFIX_PATH="path to compiler" \
  -DCMAKE_TOOLCHAIN_FILE="/absolutepath/CMSIS/DSP/armac6.cmake" \
  -DROOT="/absolutepath" \
  -DARM_CPU="cortex-m7" \
  -G "Unix Makefiles" ..

Note the absolute path used twice : for defining the ROOT (root to the CMSIS folder) and for selecting the armac6.cmake inside.

jerabaul29, 2021-03-11:
Thanks for your help, sorry for the delay.

Ok, the computer I tried to build from had CMake 3.16.3, wonder if this was contributing to the problems. I am trying to downgrade but not much success. Will try the other tips and let you know.

jerabaul29, 2021-03-11:
Ok, I am not able to downgrade to the same version as you, and I tried with the absolute paths, it did not help:

```
~/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build$ cmake -DCMAKE_PREFIX_PATH="/home/jrmet/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc" -DCMAKE_TOOLCHAIN_FILE="/home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/armac6.cmake" -DROOT="/home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5" -DARM_CPU="cortex-m4" -G "Unix Makefiles" ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_3c3d1/fast && /usr/bin/make -f CMakeFiles/cmTC_3c3d1.dir/build.make CMakeFiles/cmTC_3c3d1.dir/build
    make[1]: Entering directory '/home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_3c3d1.dir/testCCompiler.c.o
    /usr/bin/cc   -mcpu=cortex-m4 --target=arm-arm-none-eabi    -o CMakeFiles/cmTC_3c3d1.dir/testCCompiler.c.o   -c /home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeTmp/testCCompiler.c
    cc: warning: ‘-mcpu=’ is deprecated; use ‘-mtune=’ or ‘-march=’ instead
    cc: error: unrecognized command line option ‘--target=arm-arm-none-eabi’
    make[1]: *** [CMakeFiles/cmTC_3c3d1.dir/build.make:66: CMakeFiles/cmTC_3c3d1.dir/testCCompiler.c.o] Error 1
    make[1]: Leaving directory '/home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeTmp'
    make: *** [Makefile:121: cmTC_3c3d1/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "/home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeOutput.log".
See also "/home/jrmet/Desktop/Current/CMSIS-Build/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeError.log".
```

Not sure what the problem is. Any other way your would recommend building this, and / or any other way this can be distributed?

I do not have too much time to spend on this, so afraid I will just use kissfft.

christophe0606, 2021-03-11:
@jerabaul29 I think the newest cmake versions have introduced some new detection logic for the compiler and it is causing trouble.

cmake is not the official way and normally we only support the CMSIS-Pack for building.

The simplest for you may just be to compile the root C files we have in each folder like for instance : BasicMathFunctions.c for folder BasicMathFunctions. This file is including all of the other C files from the folder.

Then you need DSP/Includes DSP/PrivateIncludes and Core/Include and of course the right compilation options for the compiler.

Something like:

`armclang -DARM_MATH_LOOPUNROLL -I/cmsisroot/CMSIS/Core/Include -I/cmsisroot/CMSIS/DSP/PrivateInclude -I/cmsisroot/CMSIS/DSP/Include  -mcpu=cortex-m7 -mfpu=fpv5-d16 --target=arm-arm-none-eabi -Ofast -ffast-math -DNDEBUG -Wall -Wextra  -Werror   -mfloat-abi=hard -mlittle-endian -mthumb -c BasicMathFunctions.c`

jerabaul29, 2021-03-11:
Ok, thanks for your kind help and answer.

Unfortunately this is a bit too much work to do for me, stating to hit the "convenience vs. performance" wall compared with just using kiss-fft that works "fine enough" for my problems, and requires no work at all from me (just copy pasting a couple of files into my project folder).

I will point again the Sparkfun team to this thread and see if they can provide these functions and tools available as libraries out-of-the-box for their core, which would be the best. But if they do not, I think I will just default to kiss-fft. Maybe if you are able to discuss with @Wenn0101 about how they can compile it "once and for all" so that the binaries can be provided directly by them that would be optimal.

closed 2021-03-18 by christophe0606