ARM-software/CMSIS_5 #1139: Does CMSIS Cortex-A8 compliant ?

dehml, 2021-03-03:
The documentation say that _"Standardized API and basic run-time system for the Cortex-A5/A7/A9 processor core and peripherals."_ 
I do not see any special file for each mentioned Cortex-A processors, so does this exclude every other Cortex-A, even Cortex within the same ARMv7 architecture and so even Cortex-A8 ?
Thanks.

JonatanAntoni, 2021-03-04:
Hi @dehml,

indeed we do not have separate core headers in case of Cortex-A, but only one [core_ca.h](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core_A/Include/core_ca.h). But this header contains definitions limited to Cortex-A5/7/9. 

I expect the common stuff to apply to Cortex-A8 in a similar way. But if one wants to use CMSIS-Core(A) for Cortex-A8 some work might be required. I am happy to receive contributions.

Cheers,
Jonatan

dehml, 2021-03-04:
Alright, thanks @JonatanAntoni

closed 2021-03-04 by dehml