ARM-software/CMSIS_5 #1142: Using a shared twiddle table for different size FFTs

Alex-EEE, 2021-03-05:
Hi,

I notice on the deprecated cfft function, cfft_radix2, you support using a single twiddle table along with a twidCoefModifier to support FFTs smaller than the twiddle table size (see code below).

But on the newer versions (arm_cfft_f32 for instance), you DO NOT support using one twiddle table.  The code is written such that you must include a twiddle table for every single FFT size you want to use.

Am I missing something here?  Is there another, non deprecated version that allows you to use just one table and a twidCoefModifier?

```
  typedef struct
  {
          uint16_t fftLen;                   /**< length of the FFT. */
          uint8_t ifftFlag;                  /**< flag that selects forward (ifftFlag=0) or inverse (ifftFlag=1) transform. */
          uint8_t bitReverseFlag;            /**< flag that enables (bitReverseFlag=1) or disables (bitReverseFlag=0) bit reversal of output. */
    const float16_t *pTwiddle;               /**< points to the Twiddle factor table. */
    const uint16_t *pBitRevTable;            /**< points to the bit reversal table. */
          uint16_t twidCoefModifier;         /**< twiddle coefficient modifier that supports different size FFTs with the same twiddle factor table. */
          uint16_t bitRevFactor;             /**< bit reversal modifier that supports different size FFTs with the same bit reversal table. */
          float16_t onebyfftLen;             /**< value of 1/fftLen. */
  } arm_cfft_radix2_instance_f16;
```

christophe0606, 2021-03-08:
@Alex-EEE You are right. The cfft functions no more support a twidCoefModifier to use the same twiddle table for several different FFT lengths.

But we have introduced compilation flags to only build and include the needed twiddle tables for the FFT lengths used in the final application.

Alex-EEE, 2021-03-09:
Thanks!  I hope you all won't actually remove the cfft_radix2 functions, as I think that's a very valuable use case to be able to minimize ROM/table usage even though it might be a little slower  

One more question...what's the difference between twiddleCoef and realCoefA/realCoefB?  I see I end up with both tables when I compile, even when choosing macros to select the correct FFT size:
```

.text | .rodata | 1036598 | 32768 | realCoefBQ31 |   | bin/arm_common_tables.o | realCoefBQ31 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
.text | .rodata | 102a598 | 32768 | realCoefAQ31 |   | bin/arm_common_tables.o | realCoefAQ31 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
.text | .rodata | 1032598 | 16384 | realCoefBQ15 |   | bin/arm_common_tables.o | realCoefBQ15 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
.text | .rodata | 1026598 | 16384 | realCoefAQ15 |   | bin/arm_common_tables.o | realCoefAQ15 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
.text | .rodata | 103e658 | 384 | twiddleCoef_64_q31 |   | bin/arm_common_tables.o | twiddleCoef_64_q31 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
.text | .rodata | 103e598 | 192 | twiddleCoef_64_q15 |   | bin/arm_common_tables.o | twiddleCoef_64_q15 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  

```

christophe0606, 2021-03-09:
RFFT is based on half length CFFT. So there are twiddle tables for the CFFT. But different twiddle tables are needed to recombine those CFFT complex values into the real final ones.

Unfortunately, some of the RFFT versions are using big realCoefA and realCoefB tables even if the RFFT length is small.

If you don't need the RFFT in your code, there are some macros to avoid those big tables.

Since it may be a bit difficult to find the right combination of compilation defines, I have introduced a very experimental Python script for this in DSP/cmsisdspconfig.py 

It will create a UI in the browser and generate the options for the compiler depending on the selected options. First lines of comment in the script are explaining how to use it.

closed 2021-03-09 by Alex-EEE