ARM-software/CMSIS_5 #1147: Missing arm_sort_f32 on STM32F429 library

yjy19999, 2021-03-15:
[image]
I refered this function in the CMSIS-DSP reference manual but cannot find any declaration in the CMSIS library. Is this a vague function that did not be fully implemented or my chip do not support this feature?

christophe0606, 2021-03-16:
@yjy19999 Normally it can be found in the SupportFunctions folder of CMSIS-DSP.

Which version of the CMSIS-DSP are you using ?

yjy19999, 2021-03-16:
Thanks for your replying
The version is v1.5.3
[image]

I used the CubeMX to generate my code and the DSP is copied from the "Drivers" file in the project.
The version of CubeMX is v6.1.0
This version is fixed for me as our team adopt this version for stability.
Maybe I have to get the DSP library from the ARM website rather than CubeMX?

christophe0606, 2021-03-17:
@yjy19999  It is indeed a too old version. You need to get a more recent version to be able to build on a host. You can get the version on the development branch.

jkrech, 2021-03-17:
Note that ST has a public github repository for the STM32Cube F4 Firmware Library here:
https://github.com/STMicroelectronics/STM32CubeF4
May I ask you to raise an issue there, that they are including an outdated version of CMSIS (including DSP).
Thanks a lot.
Joachim

closed 2021-03-18 by christophe0606