ARM-software/CMSIS_5 #1149: compiling for host (MacOS) gives compilation error in SupportFunctions/arm_q7_to_q31.c

sigurdpl, 2021-03-16:
I followed the advice in #1015 for compiling on MacOS.

I am now getting this error:
/CMSIS_5-5.7.0/CMSIS/DSP/Source/SupportFunctions/arm_q7_to_q31.c:120:16: error: implicit declaration of function '__ROR' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    *pDst++ = (__ROR(in, 8)) & 0xFF000000;

cmake --build ../CMSIS_5-5.7.0/cmsis_test/cmake-build-debug --target testcmsisdsp -- -j 6

sigurdpl, 2021-03-16:
Update, i tried the Development branch. It compiles. The problem one was the 5.7.0 release

sigurdpl, 2021-03-16:
Only problem now is the SET(CMAKE_C_FLAGS_DEBUG "-D_DEBUG -D__GNUC_PYTHON__"  CACHE INTERNAL "") seems to disable debugger

christophe0606, 2021-03-17:
@sigurdpl I don't think I am touching the CMAKE_C_FLAGS when building for host. 
What's inside your cmake-build-debug ?

christophe0606, 2021-03-17:
@sigurdpl Note that there is a new option. You no more need to set the CMAKE_C_FLAGS to force __GNUC_PYTHON__  

You can just use -DHOST=ON as a cmake option for the CMSIS-DSP.

sigurdpl, 2021-03-17:
Hi, thanks for the reply. I will definitely try that. This worked though: SET(CMAKE_C_FLAGS_DEBUG "${CMAKE_C_FLAGS_DEBUG} -D__GNUC_PYTHON__"  CACHE INTERNAL "")

sigurdpl, 2021-03-17:
But will definitely try with the -DHOST=ON option instead

sigurdpl, 2021-03-17:
Removed completely the SET(CMAKE_C_FLAGS_DEBUG "${CMAKE_C_FLAGS_DEBUG} -D__GNUC_PYTHON__" CACHE INTERNAL "")
Set -DHOST=ON

That totally worked!

closed 2021-03-18 by christophe0606