ARM-software/CMSIS_5 #1150: Core: reference to CPSR in comments for M-profile intrinsics

flit, 2021-03-18:
The documentation comments for these M-profile intrinsics incorrectly reference the I- or F-bits of the CPSR register:

- `__enable_irq()`
- `__disable_irq()`
- `__enable_fault_irq()`
- `__disable_fault_irq()` 

These files are affected:

- `cmsis_armcc.h`
- `cmsis_armclang_ltm.h`
- `cmsis_armclang.h`
- `cmsis_gcc.h`

Example: https://github.com/ARM-software/CMSIS_5/blob/78ad4d7bb0cb633d7db7eb0ca9af5aabea31295a/CMSIS/Core/Include/cmsis_gcc.h#L211-L219

GuentherMartin, 2021-03-18:
Correct would be
 - Enables IRQ interrupts by clearing special-purpose register PRIMASK.
 - Disables IRQ interrupts by setting special-purpose register PRIMASK.
 - Enables FIQ interrupts by clearing special-purpose register FAULTMASK
 - Enables FIQ interrupts by setting special-purpose register FAULTMASK

JonatanAntoni, 2021-06-08:
@flit, may I ask you to double check the provided fix and close this issue, please?

flit, 2021-06-08:
💯

closed 2021-06-08 by flit