ARM-software/CMSIS_5 #1158: CFFT: Missing declaration for arm_cfft_init_f32()

philippsandi, 2021-03-22:
Hi there,
first of all: thank you very much for this library! :-)

I'm trying to compile arm_fft_bin_example_f32.c, but there's an error regarding the function arm_cfft_init_f32(), it seems to be missing. I tried to manually search for it and couldn't find it either - could you have a look at this?

Thanks and regards,
Philipp

christophe0606, 2021-04-06:
@philippsandi Can you give more info ? What is the error message you get when building ?
How are you building ? (MDK, make ? )
What is your version of CMSIS-DSP.

Thank you.

philippsandi, 2021-04-06:
Yeah sure, thanks for having a look at it. 
I'm using IAR Embedded Workbench for building. Unfortunately, I can't seem to find what version of CMSIS-DSP it's using internally.

So the error messages are:
1) warning from compiler: function "arm_cfft_init_f32" declared implicitly
2) error from linker: no definition for "arm_cfft_init_f32"

But I guess it's not very much dependant on the toolchain, as If you search for "arm_cfft_init_f32" in this repository, there seems to be no declaration in the header files anywhere.

christophe0606, 2021-04-06:
@philippsandi It is declared in arm_math.h for older cmsis-dsp version. For more recent versions, the headers were refactored and it is declared in dsp/transform_functions.h (which is now included by arm_math.h).

philippsandi, 2021-04-06:
Ah I see, now I checked arm_math.h and the file header says V1.5.3 - so it uses an older cmsis-dsp version. Unfortunately, in the local V1.5.3 arm_math.h, there's neither an include of dsp/transform_functions nor a declaration of arm_cfft_init_f32 :-D 

Well then I'll see if I can update the CMSIS-DSP that IAR is using. Thanks!

philippsandi, 2021-04-06:
Interesting, so I tried compiling the project with Segger Embedded Studio instead, because there you can see what CMSIS package it's using, and it's showing 5.07, so I expected to find V1.8.0 in arm_math.h, but instead it's V1.4.5 b, strange...

christophe0606, 2021-04-06:
@philippsandi With older cmsis-dsp, initializing the fft was done by using pointers to predefined structures (this is still working : arm_const_structs.h).

So, I am not sure if your version is containing the arm_cfft_init_f32. But you can just try by declaring it. The API is :

`  arm_status arm_cfft_init_f32(
  arm_cfft_instance_f32 * S,
  uint16_t fftLen);
`
Otherwise, just use one of the instances in arm_const_structs.h instead of arm_cfft_init_f32 in the example.

philippsandi, 2021-04-06:
Alright, thanks - I'll try to use the current cmsis-dsp instead, I think there's some misconfiguration in the IDE I'm using, so I'll try to resolve that first. Thanks for your help!

closed 2021-04-06 by philippsandi