ARM-software/CMSIS_5 #1162: Use RTX in VisualGDB to produce an exception

zhangjingru, 2021-03-25:
Use RTX in VisualGDB to produce an exception
As long as RTX is referenced in the project, the program cannot run normally, remove RTX to resume operation

Recurrence conditions
VisualGDB Build ToolChain Keil ARMCLANG V6
Keil Components -> RTOS2 -> RTX5
STM32H750VB (Only, F1 F4 normal operation)

JonatanAntoni, 2021-03-29:
Hi @zhangjingru,

May I ask you to be slightly more specfic and verbose about your issue, please?

I am not familiar with VisualGDB myself. RTX5 may cause exceptions (i.e. faults, typically hardfault) if the interrupt priorities are not configured properly. This behaviour should not depend on any specific debugger used.

Cheers,
Jonatan

zhangjingru, 2021-04-09:
> 嗨@zhangjingru，
> 
> 请允许我对您的问题再具体一点点，再详细些些吗？
> 
> 我自己不熟悉VisualGDB。如果中断优先级配置不正确，RTX5可能会导致异常（即故障，通常为硬故障）。此行为不应取决于所使用的任何特定的调试器。
> 
> 干杯，
> 乔纳丹

I recently solved this problem. The ROM startup address is not set by default in the VisualGDB project properties.
thank you for your reply

closed 2021-04-09 by zhangjingru