ARM-software/CMSIS_5 #1168: arm_convolve_HWC_q15_fast seems to only work with even dim_im_out

kwmaeng91, 2021-04-08:
arm_convolve_HWC_q15_fast tiles x-dimension into two. This is done by `if (i_out_x & 0x1)` at Line 137.

However, because there is no logic to calculate the remainder when the x dimension is odd,
from looking at the code it seems like it must produce an incorrect result when `dim_im_out `is odd.
(It won't calculate the last x or will cause segfault at Line 133 `pBuffer += ch_im_in;`, which I actually experienced)

It says in the comment that `ch_im_in `and `ch_im_out `must be a multiple of 2, and it checks this in the code, but nothing is said about `dim_im_out`. Am I doing something wrong? Or does this function only work with even `dim_im_out `but it is not specified?

felix-johnny, 2021-08-13:
@kwmaeng91 Sorry for the delay and thanks for raising this issue. I think it is case of the documentation missing a condition. 

So, for the im2col in these functions, the idea is that we fill up two columns(2*ch_im_in*dim_kernel*dim_kernel) and then operate on that.  'if (i_out_x & 0x1)'  works fine only for even dim_im_out. I can  suggest how to make it generic and help with the reviews if you implement it. 
1. The if (i_out_x & 0x1) is to be replaced by something like https://github.com/ARM-software/CMSIS_5/blob/01164235b4da4b5b7774b1336dd370261f866bce/CMSIS/NN/Source/ConvolutionFunctions/arm_convolve_s8.c#L235
2. The leftover or odd case is to be handled like in https://github.com/ARM-software/CMSIS_5/blob/01164235b4da4b5b7774b1336dd370261f866bce/CMSIS/NN/Source/ConvolutionFunctions/arm_convolve_s8.c#L256

  Let me know if you plan to contribute. In the meantime, I'll update the documentation and the check in the function so other developers don't face the issue as you did or at least don't have to debug seg faults.

felix-johnny, 2021-08-16:
I've updated the check to include the dimension of the output as well. With that, I am closing this ticket. Thanks again for raising the issue.

closed 2021-08-16 by felix-johnny

kwmaeng91, 2021-08-16:
Thank you for your response!
I would have gladly contributed if I saw your email earlier, but I guess
you already closed it.

Thank you,
Best regards,
Kiwan Maeng