ARM-software/CMSIS_5 #1169: DSP: Issues with some matrix sources

TTornblom, 2021-04-09:
While compiling the DSP libraries for IAR I ran into a few apparent bugs. Not sure how these got through for the other toolchains, but it might be a config issue.

https://github.com/ARM-software/CMSIS_5/blob/a4168d3ca5c4e2bf0cb3da7275884c97b97843a2/CMSIS/DSP/Source/MatrixFunctions/arm_mat_ldlt_f64.c#L91

This line seems bad.  "pp" is a pointer to a uint16_t, not a structure.

The same with line 383

Same here:
https://github.com/ARM-software/CMSIS_5/blob/a4168d3ca5c4e2bf0cb3da7275884c97b97843a2/CMSIS/DSP/Source/MatrixFunctions/arm_mat_ldlt_f32.c#L383

https://github.com/ARM-software/CMSIS_5/blob/a4168d3ca5c4e2bf0cb3da7275884c97b97843a2/CMSIS/DSP/Source/MatrixFunctions/arm_mat_solve_lower_triangular_f16.c#L165

No identifier "ut" here

or here:
https://github.com/ARM-software/CMSIS_5/blob/a4168d3ca5c4e2bf0cb3da7275884c97b97843a2/CMSIS/DSP/Source/MatrixFunctions/arm_mat_solve_lower_triangular_f32.c#L264

or here:
https://github.com/ARM-software/CMSIS_5/blob/a4168d3ca5c4e2bf0cb3da7275884c97b97843a2/CMSIS/DSP/Source/MatrixFunctions/arm_mat_solve_lower_triangular_f64.c#L58

JonatanAntoni, 2021-04-12:
This code seems to have more issues, e.g.

https://github.com/ARM-software/CMSIS_5/blob/a4168d3ca5c4e2bf0cb3da7275884c97b97843a2/CMSIS/DSP/Source/MatrixFunctions/arm_mat_ldlt_f64.c#L92

This looks like copy-paste-issue. Might be worth to apply some sort of static code analysis.

christophe0606, 2021-04-12:
@TTornblom  You're right. The config with ARM_MATH_SIZE_MISMATCH is generally the one I test last. And you tested it before me. There are some build issues.
I am working on it.

christophe0606, 2021-04-13:
@TTornblom It should now be corrected in commit https://github.com/ARM-software/CMSIS_5/commit/6f8bea76ff635d6f5adb5858525fc802bd2bef80

closed 2021-06-14 by christophe0606