ARM-software/CMSIS_5 #1172: core_cm4.h incorrect use of SCB_SHCSR_MEMFAULTACT_Pos in MMFSR defines

LudditeCoder, 2021-04-14:
The MMFSR related defines (lines 616 to 642 in commit fa6076b) use SCB_SHCSR_MEMFAULTACT_Pos as the base position in the 32-bit CFSR .  That should be SCB_CFSR_MEMFAULTSR_Pos.  Yes, both of these defines are equal to zero, so the resulting code is correct.  But that is not the proper use of SCB_SHCSR_MEMFAULTACT_Pos.

JonatanAntoni, 2021-04-15:
Hi @LudditeCoder, 
Thanks for pointing this out.

Is it feasible for you to raise your proposal as a pull request? If yes, would you be so kind and check other header files for similar issues, please? That'd be very helpful.

For backward compatibility reasons we should keep the old define next to a new one. If the old one is actually wrong and should not be used, it can be marked as deprecated.

Cheers,
Jonatan

GuentherMartin, 2021-06-04:
Hi @LudditeCoder, could you please check the changes and close the issue if the problem is solved.

LudditeCoder, 2021-06-05:
Yes, those look like all the changes needed.  Thank you.  And my apologies to @JonatanAntoni , with the best of intentions I kept saying "yeah, I should do that... tomorrow".

closed 2021-06-05 by LudditeCoder