ARM-software/CMSIS_5 #1174: SPI_MultiSlave CS Support

Jonny-vb, 2021-04-16:
Hi,
I appreciate that this probably isn't really the correct place for this, but I know that you guys are good at responding to issues, whereas it seems the CMSIS-Driver team can take a while to get back to issues on their repo. My hope is that you have some knowledge about the CMSIS driver layer as well. My question is as below, copied direction from my issue on their repo (https://github.com/ARM-software/CMSIS-Driver/issues/10)

Hi - I'm trying to get some code working using the SPI_MultiSlave driver. However I am having some issues with controlling the CS lines. From the documentation it looks like all I should have to do is implement SPI_Control_SlaveSelect to set the correct IO for each device, however this only seems to get called once per call to Read/Write/Transfer functions and only ever seems to set the line high.

Is there something I'm missing here, or misunderstanding? I should say I'm currently only trying to control one device through the multislave driver (to be expanding when I get the one working!).

JonatanAntoni, 2021-04-19:
Hi @Jonny-vb,

thanks for getting in touch. The team working on CMSIS-Drivers is basically the same in both repos. The CMSIS-Driver repo is a collection of driver implementations. The CMSIS-Driver API specification is part of this repo, here.

Are you trying to write your own driver implementing the SPI_MultiSlave API? If this is the case, I think you question is best sorted here. 

If you are rather using one of the driver implementations provided in the other repo, we should keep the question over there.

Cheers,
Jonatan

Jonny-vb, 2021-04-19:
Morning @JonatanAntoni 

I'm trying to use the ARM implementation for this so I'll close this issue for now, but if you could give someone a nudge over on the driver repo I'd really appreciate it!
Thanks
Jonny

RobertRostohar, 2021-05-17:
The discussion about this issue continues in the CMSIS-Driver repo as it relates to the implementation of the SPI MultiSlave driver within that repo. Therefore it should be closed here.

Jonny-vb, 2021-05-25:
The related issue in the CMSIS-Driver repo has now been noticed, so I'm happy to close this issue here again.

closed 2021-05-25 by Jonny-vb