ARM-software/CMSIS_5 #1179: Where `__TARGET_ARCH_7E_M` is defined?

SoufSilence, 2021-04-22:
When I browsed through the cmsis source code, I found the code in `cmsis_armcc.h` as follows:

```c
#if (defined (__TARGET_ARCH_7E_M) && (__TARGET_ARCH_7E_M == 1))
  #define __ARM_ARCH_7EM__          1
#endif
```
The meaning of these code is easy to understand, but I searched through all the cmsis source code files and couldn't find where `__TARGET_ARCH_7E_M` is defined. However, when I complier and debug cmsis on Arm-M4 cpu, I found `__TARGET_ARCH_7E_M` is defined and equals to 1. 
This confuses me. Can anyone explain where `__TARGET_ARCH_7E_M` is defined?

JonatanAntoni, 2021-04-22:
Hello @SoufSilence,

this is a compiler predefined macro, see https://developer.arm.com/documentation/dui0472/m/Compiler-specific-Features/Predefined-macros?lang=en.

Btw: Why are you still using discontinued Arm Compiler 5?

Cheers,
Jonatan

SoufSilence, 2021-04-22:
> 
> 
> Hello @SoufSilence,
> 
> this is a compiler predefined macro, see https://developer.arm.com/documentation/dui0472/m/Compiler-specific-Features/Predefined-macros?lang=en.
> 
> Btw: Why are you still using discontinued Arm Compiler 5?
> 
> Cheers,
> Jonatan

Hi, Jonatan: 

Thanks for your quick reply very much! You really solved my confusion. 

I'm a beginner in Arm, so I use the example provided by the MCU vendor to get started. This example uses Arm Compiler 5. I just searched the differences between the Arm Compiler 5 and 6. And it looks like Arm Compiler 6 has a lot of advantages over 5. So I will use Arm Compiler 6 to compile my programs in the future.

JonatanAntoni, 2021-06-08:
Closing due to inactivity. Please feel free to reopen if anything left to be discussed.

closed 2021-06-08 by JonatanAntoni