ARM-software/CMSIS_5 #1182: Wrong bit reversal table on arm_cfft_radix2_init_q31 and _q15

Alex-EEE, 2021-04-27:
I was getting seg faults trying to use the cfft_radix2 ( I know they're deprecated but they take way less ROM (ie can use way less tables) and sometimes are faster ).

I looked at the CMSIS 4 implementation and I noticed the table changed from armBitRevTable to armBitRevIndexTable_fixed_4096.  I think this is wrong.  I think that's a radix 4 table (or something) and so it doesn't work.  I noticed that _f32 still uses armBitRevTable.

CMSIS 4 version:
```
arm_status arm_cfft_radix2_init_q15(
  arm_cfft_radix2_instance_q15 * S,
  uint16_t fftLen,
  uint8_t ifftFlag,
  uint8_t bitReverseFlag)
{
 
  /*  Initializations of structure parameters depending on the FFT length */
  switch (S->fftLen)
  {
  case 4096u:
    /*  Initializations of structure parameters for 4096 point FFT */

    /*  Initialise the twiddle coef modifier value */
    S->twidCoefModifier = 1u;
    /*  Initialise the bit reversal table modifier */
    S->bitRevFactor = 1u;
    /*  Initialise the bit reversal table pointer */
    S->pBitRevTable = (uint16_t *) armBitRevTable;

    break;

  case 2048u:
    /*  Initializations of structure parameters for 2048 point FFT */

    /*  Initialise the twiddle coef modifier value */
    S->twidCoefModifier = 2u;
    /*  Initialise the bit reversal table modifier */
    S->bitRevFactor = 2u;
    /*  Initialise the bit reversal table pointer */
    S->pBitRevTable = (uint16_t *) & armBitRevTable[1];
```

christophe0606, 2021-04-28:
@Alex-EEE Thanks for the report.
It is indeed a bug which was introduced recently.

The function is deprecated but it should still behaves as before.

I'll make a fix and update this issue when it is done.

christophe0606, 2021-04-28:
It should be corrected in commit https://github.com/ARM-software/CMSIS_5/commit/5bc9e87a4f830093dad200ce41704c984453a74e

closed 2021-06-14 by christophe0606