ARM-software/CMSIS_5 #1187: cmsis_gcc has different definition of __disable_irq()

GorgonMeducer, 2021-04-29:
In cmsis_armclang.h and cmsis_armcc.h, the __disable_irq() will return the value of PRIMASK. But in cmsis_gcc.h the __disbale_irq() has a different behaviour that is DISABLE IRQ without return the original value of PRIMASK.

This has been there for quite a while.

GuentherMartin, 2021-04-30:
@GorgonMeducer Thanks for pointing out.
It seems to be inconsistent between the different compiler header files, but in CMSIS **___disable_irq()_** should be handled as **_void __disable_irq(void)_**.
ARMCC compiler is not able to generate code which returns PRIMASK register in any case. Please see https://developer.arm.com/documentation/dui0491/i/Compiler-specific-Features/--disable-irq-intrinsic?lang=en.
If you need the previous value of PREMASK please use function **___get_PRIMASK()_**.

GorgonMeducer, 2021-05-02:
Hi, for cortex-m,  armcc's version returns PRIMASK value when calling __disable_irq(). 
Please check it again. 

Cheers. 
Gabriel

GuentherMartin, 2021-05-03:
@GorgonMeducer Please excuse, you are of course right.
Could you please prepare a PR to change the function from **_void __disable_irq(void)_**  to **_uint32_t __disable_irq(void)_**?
Thanks in advance.

GuentherMartin, 2021-05-03:
@GorgonMeducer should function **___disable_fault_irq_** be changed accordingly?
Function **___disable_fault_irq_** is based on intrinsic **___disable_fiq_** and here the previous value of **_faultmask_** is returned.

GorgonMeducer, 2021-05-03:
Hi @GuentherMartin  
Thank you for confirming this issue. About __disable_fault_irq(), I think the key is consistency:

- consistent among supported compilers, i.e. ac5, ac6, iar and GCC
- consistent between __disable_fault_irq() and __disable_irq()

So, if you can do it for __disable_fault_irq(), that would be very nice!!!

Thank you again.

Cheers,
Gabriel

JonatanAntoni, 2021-05-04:
@TTornblom, may I ask you to add IAR's point of view to this discussion, please?

The current `cmsis_iccarm.h` headers map the CMSIS defines under discussion to certain builtins. According to the Compiler header these functions seems to be `void`.

Is it feasible to have the functions to return `PRIMASK`/`FAULTMASK` as requested?

GorgonMeducer, 2021-05-04:
@JonatanAntoni 

As long as we have **consistency**, I am ok. 
So, if you decide to use "void" for all the __disable_irq() and __disable_fault_irq(), I am also happy.

GuentherMartin, 2021-05-05:
The current usage for **___disable_irq()_** and **___disable_fault_irq()_** is **_void_**. Maybe it is not clear described in documentation.
I am going prepare a PR with reworked compiler header files.
- For IAR no changes are required as the functions are already void.
- For GCC no changes are required as the functions are already void.
- For ARMClang add inline assembly functions and skip arm_combat.h
- For ARMCC no changes are required as already void versions of the functions are available.

GorgonMeducer, 2021-05-08:
Issue solved.

closed 2021-05-08 by GorgonMeducer