ARM-software/CMSIS_5 #1192: RTX header files: request re-org for a GCC/Make based build

tobermory, 2021-05-03:
I describe a gcc/make build of the RTX(5) rtos here

https://github.com/tobermory/RTX-make-gcc.git

Near the end, I pointed out that the inclusions of RTX_Config.h:

```
#include "RTX_Config.h"
```

in ```Source/rtx_lib.h``` and ```Include/rtx_evr.h`` are not needed`.  If these redundant includes could be removed, the Config directory need not be included in a CPPFLAGS make variable.

JonatanAntoni, 2021-05-04:
Thanks @tobermory,

we will check the include chain.

> Note to RTX authors. I think RTX_Config.h is erroneously included by both Include/rtx_evr.h and Source/rtx_lib.h. Because of this, the Makefile here has to include RTX/Config in CPPFLAGS. If I remove the #include RTX_Config.h from these files, RTX will still build from source. The forced inclusion of RTX/Config in CPPFLAGS exacerbates the issue of recognizing that RTX_Config.h is really an end-user responsibility. I concede that it lives in a 'Config' directory, which highlights it as different from other .h files, but it is still referenced by the core RTX sources (which are all those except rtx_lib.c).

Cheers,
Jonatan

RobertRostohar, 2021-05-04:
Inclusion of `RTX_Confgih.h` from `rtx_lib.h` is needed. For example the definition `OS_OBJ_MEM_USAGE` from `RTX_Config.h` is used in various `rtx_*.c` modules by including it through `rtx_lib.h`.

Similar applies also to `rtx_evr.h` which uses definitions `OS_EVR_*` from `RTX_Config.h`.

tobermory, 2021-05-04:
Thanks for the info.  Yes, I see now that just because e.g. rtx_evflags.c WILL build if rtx_lib.h did not include RTX_Config.h does NOT mean that that build will be same whether RTX_Config.h is referenced or not.  The definition of e.g. OS_OBJ_MEM_USAGE, or not, affects the contents of e.g. rtx_evflags.o.

This point has actually blown apart my understanding of how to partition an RTX build. I convinced myself that all *.c except rtx_lib.c could be built 'ahead of time' and stored as a .a file. Then, at application build time, one edits RTX_Config.h for that application, and includes rtx_lib.c in the application's source file list. Given what you have pointed out, this won't work.  You can't compile most of RTX 'early' like this, since RTX_Config.h affects too many components.

I just grepped for 'OS_' in RTX/Source/*.c. The only hits outside rtx_lib.c are OS_OBJ_MEM_USAGE and OS_EVR*.  Given that I am not using Keil, losing OS_EVR is not really a loss at all. It appears that all I am losing when doing 'early build of RTX.a' is the ability to influence OS_OBJ_MEM_USAGE.

Thanks again.

RobertRostohar, 2021-05-05:
As you have already pointed out.

RTX can be prebuilt as a library (all *.c except `rtx_lib.c`) and then `rtx_lib.c` together with the configurable `RTX_Config.h` is build with the application. This is exactly how RTX 'Library' variant is exposed as a component in CMSIS pack.

The only things that are not available with the prebuilt library are Object Memory Usage and Event Recorder. In order to use also those RTX 'Source' variant is provided that is not built ahead but rather together with the application.

closed 2021-05-05 by tobermory