ARM-software/CMSIS_5 #1199: Roadmap for future release

escherstair, 2021-05-12:
I see that the latest official release is more than one year old ([5.7.0](https://github.com/ARM-software/CMSIS_5/releases/tag/5.7.0) - April 2020).
I think that in one year a lot of improvements have been done.
Can you share the roadmap for a future release?
Maybe an expected relase date too?

Thanks

JonatanAntoni, 2021-05-12:
Hi @escherstair, 

Unfortunately I cannot provide expected release date for the upcoming 5.8.0, yet. We are still working on some bits and details which we want to include. We aim to finalize the release throughout the upcoming month.

Sorry for not being concrete, here.

Cheers,
Jonatan

enes0r, 2024-07-08:
And how about 5.10? 🙂

JonatanAntoni, 2024-07-08:
Hi @enes0r,

We don't plan any further releases of CMSIS 5 as it is superseded by [CMSIS 6](https://github.com/ARM-software/CMSIS_6).

enes0r, 2024-07-08:
Thanks for the fast reply. That's a pity, after seeing 5.9.0 being released over 2 years (and 122 commits) ago I was hoping for 5.10 :) especially that we have a dependency on CMSIS_5 and would rather to have it pointing to something like 5.10.0 instead of c12fabc7f0079d0a934e9fb6e3f81bc784c4f7d4. Well, can't have everything. Cheers.

JonatanAntoni, 2024-07-08:
Please give CMSIS 6 a try; it should be widely compatible and only require adoption for very rare use cases.