ARM-software/CMSIS_5 #1202: Priority inversion

szwychtenberg94, 2021-05-20:
Base on Code Example for osMutexPrioInherit 
- https://www.keil.com/pack/doc/CMSIS_Dev/RTOS2/html/group__CMSIS__RTOS__MutexMgmt.html.

I have problem with priority inversion. One mutex with priority inheritance and second without it. There are 2 threads with diferent priority. Task with higher prio is waiting for 1s and then second task get both mutex for 2s. Task1 start waiting for mutex without inheritance. After inheritance  mutex  released, task1 priority is assigned to task2. I think it is rtx error.
Please verify this case. Code below:

```osMutexId_t inheritance_mutex_id;
osMutexId_t raw_mutex_id;
const osMutexAttr_t InheritanceMutex_attr = {
    "inheritance_mutex",
    osMutexPrioInherit,
    NULL,
    0U};
const osMutexAttr_t RawMutex_attr = {
    "raw_mutex",
    0, ////fix: osMutexPrioInherit
    NULL,
    0U};
void HighPrioThread(void *argument)
{
    osDelay(1000U);
    osMutexAcquire(raw_mutex_id, osWaitForever); // try to acquire mutex - lock by LowPrioThread
    osMutexRelease(raw_mutex_id);
    osDelay(1000U);
    while (1)
    {
        osDelay(1000U);
    }
}

void LowPrioThread(void *argument)
{
    osMutexAcquire(raw_mutex_id, osWaitForever);
    osMutexAcquire(inheritance_mutex_id, osWaitForever);
    osDelay(2000U); // block mutex for 2s
    osMutexRelease(inheritance_mutex_id);
    osMutexRelease(raw_mutex_id);
    while (1)
    {
        osDelay(2000U);
    }
}

void Init(void)
{
    const osThreadAttr_t thread_high_attr = {
        "thread_high",  // name
        0,              // attr_bits
        NULL,           //*cb_mem
        0,              // cb_size
        NULL,           //*stack_mem
        512,            // stack_size
        osPriorityHigh, // priority
        0,              // tz_module
        0               // reserved
    };
    static osThreadId_t thread_high = osThreadNew(HighPrioThread, NULL, &thread_high_attr);
    assert(thread_high);

    const osThreadAttr_t thread_low_attr = {
        "thread_low",  // name
        0,             // attr_bits
        NULL,          //*cb_mem
        0,             // cb_size
        NULL,          //*stack_mem
        512,           // stack_size
        osPriorityLow, // priority
        0,             // tz_module
        0              // reserved
    };
    static osThreadId_t thread_low = osThreadNew(LowPrioThread, NULL, &thread_low_attr);
    assert(thread_low);
    inheritance_mutex_id = osMutexNew(&InheritanceMutex_attr);
    raw_mutex_id = osMutexNew(&RawMutex_attr);
}
```

RobertRostohar, 2021-05-24:
Thanks for reporting this issue. Looks as an issue in RTX when restoring mutex owner thread priority. Should be fixed with the referenced patch.

JonatanAntoni, 2021-06-08:
@szwychtenberg94, may I ask you to verify the fix works for you and close this issue, please?

szwychtenberg94, 2021-06-16:
Thanks for fix, it works. When is the new pack planned to be released as official?

closed 2021-06-16 by szwychtenberg94

JonatanAntoni, 2021-06-16:
@szwychtenberg94,

I am in the process of finalizing the next release. Please grant me a couple of more weeks.

Thanks,
Jonatan