ARM-software/CMSIS_5 #1204: It is possible to know used global dynamic memory from API

juliobc, 2021-05-27:
Hi, 

I need check the used global dynamic memory in real time. Is there any function in the API for that?  if not, any workaround?

Thanks

JonatanAntoni, 2021-05-27:
Hi @juliobc,

What are you exactly looking for, i.e. what do you mean by "check"?
Are you interested in how much global dynamic memory is used or free? Or anything further?

Cheers,
Jonatan

juliobc, 2021-05-28:
Hi @JonatanAntoni 

yes, I am looking for how much global dynamic memory is used or free

thanks

RobertRostohar, 2021-05-31:
There is no API to check the global dynamic memory usage. However you can check the internal structure in order to get this information.

```
#include "rtx_os.h"

uint32_t global_dynamic_memory_size;
uint32_t global_dynamic_memory_used;

global_dynamic_memory_size = *((uint32_t *)osRtxConfig.mem.common_addr + 0);
global_dynamic_memory_used = *((uint32_t *)osRtxConfig.mem.common_addr + 1);
```

juliobc, 2021-05-31:
Hi @RobertRostohar 

Thank you very much

closed 2021-05-31 by juliobc