ARM-software/CMSIS_5 #1206: Building only CMSIS-NN Library

uu-praveeng, 2021-06-03:
Hi,

I have seen the build steps for building CMSIS-DSP library at "Building only the CMSIS-DSP library" [https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/README.md](url) . I assume it takes care of DSP library building.
Is there a documented way to build CMSIS-NN Library as well? or Following similar steps in CMSIS DSP  library replacing DSP with NN will work out?
By any chance, did you try building CMSIS-NN library?

Thanks in advance.
Praveen.

felix-johnny, 2021-07-07:
@uu-praveeng  Thanks for the question. There is a way and it is linked to how the unit tests are built in CMSIS-NN. If you follow the first 4 lines of the instructions in /home/feltho01/box/cmsis/mirror-cmsis/CMSIS/NN/Tests/UnitTest/ethos-u-core-platform starting from 'mkdir build' to 'make' you get individual library files in the respective source folders like the image below.

[image]

That said, one thing to note is that it is only the *_s8 functions that are built which is in line with support for TFLM based kernels. An example of that is in https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/NN/Source/ConvolutionFunctions/CMakeLists.txt#L21

uu-praveeng, 2021-08-03:
Thanks Felix for reply. I am looking out for building all .c files into a .a file, which I could port into an IDE and deploy.
I have few open questions. 

1. Will building of DSP [module](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/README.md) include building of NN module ? 
_I tried to build DSP module as per the instructions at [link ](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/README.md) but facing issue similar to [this ](https://github.com/ARM-software/CMSIS_5/issues/1138)._ 

2. Does the pack creation include process of library building ? 
 _If yes, could you point me to sample pdsc file to do the same_

felix-johnny, 2021-08-17:
1. No, the DSP module does not build NN. The instructions for building NN as library will be up shortly.
2. It is the pdsc file in the root directory of CMSIS

felix-johnny, 2021-08-18:
@uu-praveeng Could you try out the linked PR and provide your feedback?

uu-praveeng, 2021-08-18:
@felix-johnny . Could you provide patch for the changelist? I manually merged files and tried it on windows . I am seeing below 
error   

**C:\Users\pgajula\CMSIS_5\CMSIS\NN\build>cmake -DCMAKE_TOOLCHAIN_FILE="X:\share\apps\RTT\GNUArmEmbeddedToolchain\v9_3_1\win\bin\arm-none-eabi-gcc" -DCMAKE_PREFIX_PATH="X:\share\apps\RTT\GNUArmEmbeddedToolchain\v9_3_1\win\bin" -DTARGET_CPU="cortex-m3" -G "Unix Makefiles" ..
CMake Error at C:/Program Files/CMake/share/cmake-3.20/Modules/CMakeDetermineSystem.cmake:129 (message):
  Could not find toolchain file:
  X:\share\apps\RTT\GNUArmEmbeddedToolchain\v9_3_1\win\bin\arm-none-eabi-gcc
Call Stack (most recent call first):
  CMakeLists.txt:21 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!**

However I have written a standalone Makefile which is working fine with same toolchain.Attached Makefile for your reference
[Makefile_Win.txt](https://github.com/ARM-software/CMSIS_5/files/7008677/Makefile_Win.txt)

mansnils, 2021-08-19:
@uu-praveeng I am not sure what you mean by changelist? Could you try the toolchain-cmake-file from Arm Ethos-U Core Platform?
For example:
git clone "https://review.mlplatform.org/ml/ethos-u/ethos-u-core-platform"
cmake -DCMAKE_TOOLCHAIN_FILE="X:\path\to\cloned\project\core_platform\cmake\toolchain\arm-none-eabi-gcc.cmake" -DTARGET_CPU="cortex-m3" -G "Unix Makefiles" ..

Also please check this readme: "X:\path\to\cloned\project\core_platform\README_WINDOWS.md"