ARM-software/CMSIS_5 #1210: Core: cmsis_gcc.h missing v8.1-M support in ifdefs

flit, 2021-06-08:
The `cmsis_gcc.h` header is missing the `__ARM_ARCH_8_1M_MAIN__` macro in in multiple ifdefs that include certain intrinsics by architecture version.

Example:
https://github.com/ARM-software/CMSIS_5/blob/4ed9008313cda609d698589fb72c4c979536ccf6/CMSIS/Core/Include/cmsis_gcc.h#L518-L520

Similar ifdef from `cmsis_armclang.h` for comparison:
https://github.com/ARM-software/CMSIS_5/blob/4ed9008313cda609d698589fb72c4c979536ccf6/CMSIS/Core/Include/cmsis_armclang.h#L468-L471

These are all the instances of the ifdefs in `cmsis_gcc.h` that test `__ARM_ARCH_8M_MAIN__` and are therefore likely to need fixing:

```
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:520:      (defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1))    )
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:664:            (defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1))    ) */
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:667: #if ((defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)) || \
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:681: #if (!(defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)) && \
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:703: #if (!(defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)))
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:726: #if (!(defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)) && \
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:747: #if (!(defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)))
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:768: #if (!(defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)) && \
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:791: #if (!(defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)))
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:814: #if (!(defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)) && \
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:835: #if (!(defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)))
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:844: #endif /* ((defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)) || \
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:1074:      (defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1))    )
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:1119:      (defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)) || \
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:1242:            (defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)) || \
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:1248:      (defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1))    )
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:1393:            (defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1))    ) */
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:1446:            (defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1))    ) */
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:1449: #if ((defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)) || \
/Volumes/projects/CMSIS_5/CMSIS/Core/Include/cmsis_gcc.h:1627: #endif /* ((defined (__ARM_ARCH_8M_MAIN__ ) && (__ARM_ARCH_8M_MAIN__ == 1)) || \
```

I have _not_ checked for missing v8.1-M intrinsics. (That would be a separate ticket, anyway.)

GuentherMartin, 2021-06-09:
Hello @flit,
the problem is that the used version of GCC `gcc version 10.2.1 20201103 (release) (GNU Arm Embedded Toolchain 10-2020-q4-major)` does not generate the predefined macro `__ARM_ARCH_8_1M_MAIN__` for ARM-v8.1-M cores. GCC only generates  `__ARM_ARCH_8M_MAIN__`. 
As soon as GCC generates `__ARM_ARCH_8_1M_MAIN__` it will be added.

To check the predefined macros I use the following batch file:
```
SET GCC="C:\_Tools\gcc-arm-none-eabi-10-2020-q4-major\bin\arm-none-eabi-gcc.exe"

%GCC% -mthumb -march=armv6-m                    -dM -E -x c _x.c > _macro_armv6-m.txt
%GCC% -mthumb -march=armv7-m                    -dM -E -x c _x.c > _macro_armv7-m.txt
%GCC% -mthumb -march=armv7e-m                   -dM -E -x c _x.c > _macro_armv7e-m.txt
%GCC% -mthumb -march=armv8-m.main               -dM -E -x c _x.c > _macro_armv8-m_main.txt
%GCC% -mthumb -march=armv8-m.main+dsp           -dM -E -x c _x.c > _macro_armv8-m_main_dsp.txt
%GCC% -mthumb -march=armv8-m.base               -dM -E -x c _x.c > _macro_armv8-m_base.txt
%GCC% -mthumb -march=armv8.1-m.main             -dM -E -x c _x.c > _macro_armv81-m_main.txt
%GCC% -mthumb -march=armv8.1-m.main+dsp         -dM -E -x c _x.c > _macro_armv81-m_main_dsp.txt
                                                
%GCC% -mthumb -mcpu=cortex-m0                   -dM -E -x c _x.c > _macro_cortex-m0.txt
%GCC% -mthumb -mcpu=cortex-m0plus               -dM -E -x c _x.c > _macro_cortex-m0plus.txt
%GCC% -mthumb -mcpu=cortex-m3                   -dM -E -x c _x.c > _macro_cortex-m3.txt
%GCC% -mthumb -mcpu=cortex-m4                   -dM -E -x c _x.c > _macro_cortex-m4.txt
%GCC% -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -dM -E -x c _x.c > _macro_cortex-m4f.txt
%GCC% -mthumb -mcpu=cortex-m7                   -dM -E -x c _x.c > _macro_cortex-m7.txt
%GCC% -mthumb -mcpu=cortex-m23                  -dM -E -x c _x.c > _macro_cortex-m23.txt
%GCC% -mthumb -mcpu=cortex-m33                  -dM -E -x c _x.c > _macro_cortex-m33.txt
%GCC% -mthumb -mcpu=cortex-m35p                 -dM -E -x c _x.c > _macro_cortex-m35p.txt
%GCC% -mthumb -mcpu=cortex-m55                  -dM -E -x c _x.c > _macro_cortex-m55.txt
%GCC% -mthumb -mcpu=cortex-m55 -mfloat-abi=hard -dM -E -x c _x.c > _macro_cortex-m55_fp.txt
```

flit, 2021-06-09:
Ah, that's annoying! Have you confirmed with the GNU-RM team that `__ARM_ARCH_8_1M_MAIN__` will be added in the next release? Also, do you know if there is a bug in the gcc tracker? Thanks.

GuentherMartin, 2021-06-09:
I did not talk to the GNU-RM team about __ARM_ARCH_8_1M_MAIN__.
Who should I contact about this this problem?
Please send an answer to my Arm mail account.

finchky, 2023-08-01:
Have there been any updates regarding cmsis_gcc.h adding `__ARM_ARCH_8_1M_MAIN__` support?

GuentherMartin, 2023-08-03:
Please see also https://github.com/ARM-software/CMSIS_6/pull/33